The-EG/OctoPrint-CameraSettings #1: Pre-release feedback

The-EG, 2021-04-15:
Hopefully you've found this issue because you'd like to test this plugin before it's officially released 😃.

If so, you can install it by using the 'main' URL in the OctoPrint plugin manager:
https://github.com/The-EG/OctoPrint-Camerasettings/archive/main.zip
[image]

After install and restart, you should get a new tab, 'Camera Settings.' A few settings are also available in the plugin settings, most notably the 'load preset on startup' option.

Some comments about the plugin right now:
Unfortunately, not all cameras show up in `/dev/v4l/by-id` (the RaspiCam is one), so using those id's isn't possible. Even the name may not be possible because it's possible for multiple `/dev/videoN` devices to have the same name. I'm still digging into this, but for now everything is tied to `/dev/videoN` devices, including the presets.

To have the settings applied at OctoPrint startup, you must create a preset and both enable the option in settings and select the applicable preset.

Some issues I already know about:
 - The camera preview is *massive* if you are using a plugin that causes the tabs to be 'widescreen.'
 - Some cameras won't allow some settings to be changed until other settings are set to a particular value. I'm working on a solution to this, but if you know your camera has this behavior that would be helpful to know, especially individual control names and dependencies.

Feedback needed:
 - General thoughts about how it works
 - Issues created for any cameras that display the 'camera controls have not been implemented warning.' Just click on the link.

Zardoz66, 2021-04-15:
{
  "backlight_compensation": {
    "type": "int",
    "min": "0",
    "max": "10",
    "step": "1",
    "default": "0",
    "value": "5"
  },
  "pan_absolute": {
    "type": "int",
    "min": "-201600",
    "max": "201600",
    "step": "3600",
    "default": "0",
    "value": "0"
  },
  "tilt_absolute": {
    "type": "int",
    "min": "-201600",
    "max": "201600",
    "step": "3600",
    "default": "0",
    "value": "0"
  },
  "focus_absolute": {
    "type": "int",
    "min": "0",
    "max": "40",
    "step": "1",
    "default": "0",
    "value": "19",
    "flags": "inactive"
  },
  "focus_auto": {
    "type": "bool",
    "default": "0",
    "value": "1"
  },
  "zoom_absolute": {
    "type": "int",
    "min": "0",
    "max": "10",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

jneilliii, 2021-04-15:
{
  "hue": {
    "type": "int",
    "min": "-2000",
    "max": "2000",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "gamma": {
    "type": "int",
    "min": "100",
    "max": "300",
    "step": "1",
    "default": "100",
    "value": "100"
  },
  "backlight_compensation": {
    "type": "int",
    "min": "0",
    "max": "2",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

The-EG, 2021-04-15:
Ok, these should be available now. Reinstall (no uninstall necessary) from the same URL: https://github.com/The-EG/OctoPrint-CameraSettings/archive/main.zip

@Zardoz66 You should now have an extra checkbox for 'Focus Auto' and sliders for Focus, Backlight Compensation, Pan, Tilt, & Zoom

@jneilliii You should also now have sliders for Backlight Compensation, Hue and Gamma

The-EG, 2021-04-16:
I've moved the settings from a dedicated tab to the plugin settings. If you've already installed and want to get the updated version, just reinstall it from the same URL: https://github.com/The-EG/OctoPrint-CameraSettings/archive/main.zip

Zardoz66, 2021-04-16:
Focus option will not enable/disable. always on (checked) Microsoft lifecam Cinema 
attached octopi.log camera settings filtered.
[octopilog_camsettings.txt](https://github.com/The-EG/OctoPrint-CameraSettings/files/6326208/octopilog_camsettings.txt)

closed 2021-04-29 by The-EG