The-EG/OctoPrint-CameraSettings #2: [FR] Implement Controls for [Camera Name]

jneilliii, 2021-04-15:
**Camera manufacturer and model (ie. NexiGO N660)**:
Hercules HD Twist

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "hue": {
    "type": "int",
    "min": "-2000",
    "max": "2000",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "gamma": {
    "type": "int",
    "min": "100",
    "max": "300",
    "step": "1",
    "default": "100",
    "value": "100"
  },
  "backlight_compensation": {
    "type": "int",
    "min": "0",
    "max": "2",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

The-EG, 2021-04-15:
Should be fixed in 05754d86bf4b10c5ff9675b74cdef0469634e979 !

Let me know if not.

jneilliii, 2021-04-15:
Confirmed, fixed.

closed 2021-04-15 by jneilliii