The-EG/OctoPrint-CameraSettings #3: RaspiCam 2.1 - Controls do not affect camera stream

The-EG, 2021-04-16:
User reported that controls do not have an effect on the camera stream using a Raspicam 2.1.

Normal OctoPi image, raspicam in 'raspi' mode in octopi.txt. Using 'usb' mode results in `select() timeout` in mjpg-streamer startup.

`v4l2-ctl --list-devices` and `v4l2-ctl --list-ctrls` etc all work as expected. `--list-devices` shows the raspicam as /dev/video0

The-EG, 2021-04-16:
I was able to reproduce part of this with my raspicam 1.3. It appears that running mjpg-streamer in 'raspi' mode isn't compatible.

Now, we just need to figure out how to get it into usb mode for this user.

The-EG, 2021-04-18:
Solved! User was able to use raspicam in USB mode by making the following /boot/octopi.txt:

```text
# change camera to usb
Camera=usb"

# change usb camera options to appropriate resolution and framerate
camera_usb_options="-r 1920x1080 -f 30"

# Enable the web controls
camera_http_webroot="./www"
camera_http_options=""

# comment out any raspi_options
#camera_raspi_options="-fps 10"
```

I'm not sure if enabling web controls is strictly necessary, but it didn't seem to hurt.

closed 2021-04-18 by The-EG

JonTizz, 2021-09-24:
Just is case it helps anyone (or me) in the future, I'll add what I found when doing this.

These settings worked for me, but in USB mode the resolution was causing an issue (1640x1232) and my stream never started. Might be because I'm on wifi but not 100% sure. When downgrading to 640x480 the stream showed up.

The resolutions that worked were : 

- 640x480
- 1280x720
- 1640x922
- 1920x1080 (which needed bitrate adjustment to look reasonable) 

and these which did not.

- 1640x1232
- 3280x2464

(Resolutions sourced from [the picam docs](https://picamera.readthedocs.io/en/release-1.12/fov.html#camera-modes).)

I ended up settling on 1640x922 as it had a wider FoV than 1920x1080. Will need to do more research on why `usb` mode doesn't seem to work with these resolutions, while `raspi` mode does.

For reference, my system is:
```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.82 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903988736
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.4
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin E3V2-Extensible-UI (Jun  6 2021 13:54:38)
systeminfo.generator: systemapi
```

JonnyTizz, 2021-09-24:
Just is case it helps anyone (or me) in the future, I'll add what I found when doing this.

These settings worked for me, but in USB mode the resolution was causing an issue (1640x1232) and my stream never started. Might be because I'm on wifi but not 100% sure. When downgrading to 640x480 the stream showed up.

The resolutions that worked were : 

- 640x480
- 1280x720
- 1640x922
- 1920x1080 (which needed bitrate adjustment to look reasonable) 

and these which did not.

- 1640x1232
- 3280x2464

(Resolutions sourced from [the picam docs](https://picamera.readthedocs.io/en/release-1.12/fov.html#camera-modes).)

I ended up settling on 1640x922 as it had a wider FoV than 1920x1080. Will need to do more research on why `usb` mode doesn't seem to work with these resolutions, while `raspi` mode does.

For reference, my system is:
```
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/93.0.4577.82 Safari/537.36
connectivity.connection_check: 1.1.1.1:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903988736
env.os.bits: 32
env.os.id: linux
env.os.platform: linux
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.4
env.plugins.pi_support.octopi_version: 0.18.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 20.3.3
env.python.version: 3.7.3
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.6.1
printer.firmware: Marlin E3V2-Extensible-UI (Jun  6 2021 13:54:38)
systeminfo.generator: systemapi
```

The-EG, 2021-09-24:
> Will need to do more research on why usb mode doesn't seem to work with these resolutions, while raspi mode does.

@JonnyTizz USB mode has options that can be set, and the defaults are fairly conservative. One of those is the fact that it will put the camera in 'photo' mode above a certain resolution, even if video is requested. This tanks the framerate or even keeps video from working at all at the maximum resolution.

To have the camera in video mode all the way up to maximum, create a file at `/etc/modprobe.d/raspicam.conf` with this line in it:
```
options bcm2835-v4l2 max_video_width=3280 max_video_height=2464
```

And then reboot. You'll still get limited FPS, but I'm able to run my 1.3 pi cam at maximum res at 10 FPS without issues.

If after doing that you have issues that look like low bitrate, even after setting it to maximum, you may need to increase the GPU memory to 256 (`sudo raspi-config`: Performance->GPU Memory)

JonnyTizz, 2021-09-25:
@The-EG woh, well that exactly solved the issue! I can now run on 1640x1232 at 30fps just fine, in USB mode! Thank you so much.

I'll admit to not testing the full 3280x2464, but following your steps exactly enabled me to see the stream at higher resolutions. I also blindly upped the GPU memory, because more = more better, right? 😬 

Thank you, again!