The-EG/OctoPrint-CameraSettings #6: [UI] Settings do not scroll independently of camera preview

The-EG, 2021-04-18:
This may be due to UI Customizer:
[image]

The-EG, 2021-04-20:
This is an issue with UI Customizer.

LazeMSS, 2021-04-25:
Just a couple of notes - this should be removed:

https://github.com/The-EG/OctoPrint-CameraSettings/blob/abd806f8487a2b4259467d1c638684618f49b3ad/octoprint_CameraSettings/templates/camerasettings_settings.jinja2#L3
```
{% assets "js_client" %}
<script src="{{ url_for("static", filename="js/lib/jquery/jquery.min.js") }}"></script>
<script src="{{ url_for("static", filename="js/lib/lodash.min.js") }}"></script>
<script type="text/javascript" src="{{ url_for("static", filename="webassets/packed_client.js") }}"></script>
{% endassets %}
```

The-EG, 2021-04-25:
> Just a couple of notes - this should be removed:
> 
> https://github.com/The-EG/OctoPrint-CameraSettings/blob/abd806f8487a2b4259467d1c638684618f49b3ad/octoprint_CameraSettings/templates/camerasettings_settings.jinja2#L3

Thanks! Yes, I removed that when we were talking about it the other day (and also from my other plugin where it go copied from): 44c06502227850efdf2e448b8b84ff387c475990

The-EG, 2021-04-25:
Should be fixed or at least improved in UI Customizer now.

closed 2021-04-25 by The-EG