The-EG/OctoPrint-CameraSettings #7: [FR] Implement Controls for [PLAYSTATION Eye (PS3)]

Rufus242, 2021-04-20:
**Camera manufacturer and model (ie. NexiGO N660)**:
PLAYSTATION Eye (for PLAYSTATION 3)

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "white_balance_automatic": {
    "type": "bool",
    "default": "1",
    "value": "1"
  },
  "exposure": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "1",
    "default": "120",
    "value": "127"
  },
  "gain_automatic": {
    "type": "bool",
    "default": "1",
    "value": "1",
    "flags": "update"
  }
}

The-EG, 2021-04-20:
Hey there, I've added 'White Balance Automatic', 'Exposure' and 'Gain Automatic' to the controls on the RC branch.

You can try this out now by reinstalling (no uninstall necessary) from this URL: https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/rc.zip

Rufus242, 2021-04-23:
Hi, 
the settings are working properly :)
But it seems that these new settings are not saved in the Preset profiles?

The-EG, 2021-04-23:
Ok, thank you for checking.

The settings should automatically be included when saving a new preset, but any preset saved previously wouldn't have them. And, while thinking about explaining this, you may have found a nuance with how presets are saved and loaded:

It is possible that the new settings aren't saved in a preset in a very specific situation:
 - Old preset exists before the settings were added
 - You updated to the version with the added settings
 - Deleted the old preset and saved a new one with the same name
 - Did **not** click save, but tried to load the 'new' preset

If that's the case, that is because presets are not actually 'saved' until the save button on the settings window is pressed. That could be confusing, so I'll have to look into a way to work around that.

closed 2021-04-23 by The-EG