The-EG/OctoPrint-CameraSettings #11: [FR] Implement Controls for Logitech Quickcam

newrecycle, 2021-04-26:
Logitech Quickcam 1.3mp (very old lmao)

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "privacy": {
    "type": "bool",
    "default": "0",
    "value": "0"
  }
}

I believe this is for the privacy shield over the webcam but putting the shield down has no effect other than the webcam feed being black (duh)

The-EG, 2021-04-26:
Interesting.

I'll add it in any case so the warning goes away for you and any other users who may be using a similar camera.

I'm also thinking about adding options for hiding settings that aren't important or useful, so this would probably be a good default for that.

Thanks!

The-EG, 2021-04-26:
It's now in the RC branch. If you are curious and want to see if it does anything (or just to get rid of the warning), you can install the branch with this URL (just re-install over the existing plugin): https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/rc.zip

I plan on officially releasing soon, so this will be included then, and you'll still be prompted to upgrade when that happens, etc.

newrecycle, 2021-04-27:
Wow, thanks for the quick reply and commit! I can tell you that when the privacy shield is down, the checkbox will become checked and when you uncheck it, it quickly becomes checked again, vise versa if the privacy shield is up. my suggestion is just to add a warning to the user that the privacy shield may be down, but the warning can be ignored. actually I think ill try to make that commit now

newrecycle, 2021-04-27:
fixed with PR #13

closed 2021-04-27 by newrecycle