The-EG/OctoPrint-CameraSettings #12: [BUG] New Issue

bergdahl, 2021-04-26:
**Observed behavior**
The autofocus checkbox can't be unchecked.

**Expected behavior**
It should be unchecked.

**OctoPrint Version**
1.5.3

**CameraSettings Plugin Version**
0.1.0.rc5

**Camera**
Logitech C922 Pro

**Linux/OctoPi**
Version 0.17.0, running on Raspberry Pi 4 Model B Rev 1.1

**Other Plugins**
I run  Octolapse.

[octoprint (4).log](https://github.com/The-EG/OctoPrint-CameraSettings/files/6380714/octoprint.4.log)

bergdahl, 2021-04-26:
Here are my settings.
[image]
[image]

The-EG, 2021-04-26:
from discord:
> OK, I logged into the Pi and ran some commands. According to this I have disabled the autofocus. It is still checked in the plugin settings though. 
```terminal
pi@ultimaker:~ $ v4l2-ctl -l
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=138
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=130
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=132
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=0
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=0
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=1
      white_balance_temperature 0x0098091a (int)    : min=2000 max=6500 step=1 default=4000 value=4690
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=206
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=1
              exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=183
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=0
                     focus_auto 0x009a090c (bool)   : default=1 value=0
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=100 value=100
```

The-EG, 2021-04-29:
I've added some debugging logging to version 0.1.0. Please update and enable debug logging for the plugin:
Go to settings->logging and add an entry for this plugin:
[image]

Then try to set the settings again and grab the octoprin.log for me. there should be quite a bit from the plugin in there this time.

The-EG, 2021-04-30:
I've seen a couple of users with Logitech C92x cameras say they have this issue now...if you are one and come across this issue, setup the debug logging as above and provide a log! I'm not sure if this is something I can fix or not, but I will need more info on it.

bergdahl, 2021-04-30:
Updated to 0.1.0 and played around in the settings. Here is the log. 

```2021-04-30 09:54:12,116 - octoprint.plugins.camerasettings - WARNING - Error running v42l-ctl:
exposure_absolute: Invalid or incomplete multibyte or wide character
VIDIOC_QUERYCAP: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_CTRL: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid or incomplete multibyte or wide character
```

[octoprint (8).log](https://github.com/The-EG/OctoPrint-CameraSettings/files/6403770/octoprint.8.log)

The-EG, 2021-04-30:
Hmmm ok, so it's not liking exposure absolute... that's actually not that abnormal, it can error out like that when the control isn't applicable (in this case, when exposure is in an 'auto' mode), but I have to wonder if it's getting that error and then not trying to set the other controls 'after' it or something.

ruedli, 2021-05-01:
Just for your info: the Octolapse app also has a camera settings tab: if you select its settings, an icon appears where you can do the settings. For this one the switching auto-focus on/off DOES work. You can also focus a C920 in Octolapse

The-EG, 2021-05-01:
@ruedli Good to know that OctoLapse's method works, since it's very similar. It's actually using mjpg-streamer's webcam settings webpage, so it relies on mjpg-streamer and that setup, whereas this plugin does not (it can be used with uStreamer, etc).

I have a theory that sending all of the settings to the camera at the same time may be part of the issue. I'm going to see if sending them one at a time helps.

metaColin, 2021-05-01:
I am experiencing the same issue with an AUKEY FHD Webcam (PC-LM1E).

The 'Focus, Auto' button cannot be unchecked...

The-EG, 2021-05-01:
> I have a theory that sending all of the settings to the camera at the same time may be part of the issue. I'm going to see if sending them one at a time helps.

This is now in the `fixes/c922_auto_focus` branch. You can test this out by installing the plugin from this URL:
https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fixes/c922_auto_focus.zip

No uninstall necessary, just install it over the top of the existing:
[image]

If you try this, please let me know the result! If this works I'll still need to work out how to load the presets on startup...

metaColin, 2021-05-01:
Another wrinkle: This settings page seems to fight with the camera settings page accessible via Octolapse.

Is this plugin compatible with Octolapse? I sure hope so because the camera settings in Octolapse are not very functional.

The-EG, 2021-05-01:
> Is this plugin compatible with Octolapse? I sure hope so because the camera settings in Octolapse are not very functional.

Yes, this should work fine with OctoLapse, but please disable OctoLapse's 'Custom Image Preferences' in your camera profile:
Click the 'pencil':
[image]

metaColin, 2021-05-01:
Well, that stopped them fighting over settings, but now my auto-focus is locked 'on' again...

bergdahl, 2021-05-01:
This seems to have resolved my issues. All controls now work as expected.

The-EG, 2021-05-01:
> This seems to have resolved my issues. All controls now work as expected.

[image]

That was a shot in the dark, but if it works I'll take it!

I'll wait for a few other testers if I can, then push it to an RC and try to figure out the best approach for loading settings from a preset too...

metaColin, 2021-05-01:
After further research, it seems like the Aukey FHD (PC-LM1E) webcam I am using likely has a fixed focus. Is there any way for this plugin to query the attached camera hardware and accurately mark/disable the parameters which can/cannot be adjusted?

Related question: Is there a camera that is the *most* compatible with this plugin?

The-EG, 2021-05-01:
> Is there any way for this plugin to query the attached camera hardware and accurately mark/disable the parameters which can/cannot be adjusted?

Unfortunately, that is exactly what's already being done. The camera reports back that it has a certain controls, so those are shown while anything else is hidden. In a few rare cases, these controls may not be controllable via software, but actually be indicating something on the hardware level, like if a privacy shield is down.

It may be worthwhile to put a list of cameras in the FAQ that are known to return invalid settings, and I could possibly hard code some of those to be hidden based on the camera name, but there's no way for me to know if a setting is actually valid or not.

Additionally, one new feature that I may implement soon is the ability to configure what settings are hidden, so that a user can hide those settings that they don't want to use.

ruedli, 2021-05-01:
The zip file provided in fixes/c922_auto_focus resolved the auto focus "stuck-to-on" issue for me, looks like a sound RC version to me, well done!

(not targeted, but mentioned: multicam-not-previewing: #28 still remained).

The-EG, 2021-05-01:
Now available in release candidate 0.1.2rc1

commit f6188127 (2021-05-01): Only send updated control (fix #12)
closed 2021-05-03 by The-EG