The-EG/OctoPrint-CameraSettings #14: [BUG] Afterinstall/restart, no indication there is a new plugin

Navyguy330, 2021-04-30:
**Observed behavior**
NO plugin called camera settings is available on any screen after install

**Expected behavior**
to control the camera

**OctoPrint Version**
not sure but updated like 2 days ago

**CameraSettings Plugin Version**
the current one

**Camera**
logitec c920s

**Linux/OctoPi**
OctoPi most recent version

**Other Plugins**
 I do have spaghetti detective,  I assumed it was just not going to push the same camera settings to the detective stream,. not completely make it unusable, that's a HIUGE let down.

The-EG, 2021-04-30:
Hi there, what version of python is octoprint running on?

If you don't know, open the octoprint settings window, click the system info button at the bottom, click the copy button on that window and paste that here.

Navyguy330, 2021-04-30:
3.7.3

The-EG, 2021-04-30:
Ok. Please provide octoprint.log

Navyguy330, 2021-04-30:
I know you mentioned incompatible with detective.  I really just thought it would not have any effect on the detectives stream rather than prevention of use for when I am using the octopi locally. (assuming due to the level at which one or both of these utilizes the stream, from a non programmers stand point if you get my thoughts) 1 sec on log. idk if its logging anything?

Navyguy330, 2021-04-30:
not sure how to the the log?

The-EG, 2021-04-30:
> I know you mentioned incompatible with detective. I really just thought it would not have any effect on the detectives stream rather than prevention of use for when I am using the octopi locally. [...]

Unfortunately, TSD premium completely takes over the streaming so that it can be converted to a format more appropriate for transmission over the internet. The method that it uses to do this means that it has complete control over the camera, meaning anything else on the system can do anything with it. 

But in any case, that shouldn't have prevented the plugin from loading, it just wouldn't function. The settings would still be available under the plugin settings (it does not create a new tab).

>not sure how to the the log?

To get the log, open the settings window, go to Logging and click the download button next to octoprint.log and then attach it here. At the very least, that will allow me to see if there's a reason it didn't load, etc.

Navyguy330, 2021-04-30:
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/6402622/octoprint.log)

The-EG, 2021-04-30:
It is loaded:
> 2021-04-30 02:29:50,530 - octoprint.plugin.core - INFO - 27 plugin(s) registered with the system:
|  Access Anywhere - The Spaghetti Detective (1.6.6) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_thespaghettidetective
|  Action Command Notification Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_notification
|  Action Command Prompt Support (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/action_command_prompt
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/announcements
|  Anonymous Usage Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/tracking
|  Application Keys Plugin (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/appkeys
|  Backup & Restore (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/backup
|  Bed Visualizer (1.0.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_bedlevelvisualizer
**|  Camera Settings (0.1.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_CameraSettings**
|  Cancel Objects (0.4.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_cancelobject
|  Core Wizard (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/corewizard
|  Discovery (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/discovery
|  Error Tracking (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/errortracking
|  Exclude Region (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_excluderegion
|  File Check (2021.2.23) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_file_check
|  Firmware Check (2021.2.4) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_firmware_check
|  GCode Viewer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/gcodeviewer
|  GcodeEditor (0.2.12) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_GcodeEditor
|  Logging (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/logging
|  Octolapse (0.4.1) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_octolapse
|  Pi Support Plugin (2021.3.26.post2) (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_pi_support
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/pluginmanager
|  PrintTimeGenius Plugin (2.2.7) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_PrintTimeGenius
|  Resource Monitor (0.3.3) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_resource_monitor
|  Software Update (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/softwareupdate
|  SpoolManager Plugin (1.3.4) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_SpoolManager
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/virtual_printer
Prefix legend: ! = disabled, # = blacklisted, * = incompatible
2021-04-30 02:29:50,564 - octoprint.environment - INFO - Detected environment is Python 3.7.3 under Linux (linux). Details:

and I can see that it went looking for cameras when you opened the settings window:

> 2021-04-30 02:30:50,981 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture0 based on ^bcm2835-isp-\w+$
2021-04-30 02:30:50,982 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-isp based on ^bcm2835-codec-\w+$
2021-04-30 02:30:51,022 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-decode based on ^bcm2835-codec-\w+$
2021-04-30 02:30:51,024 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture1 based on ^bcm2835-isp-\w+$
...
2021-04-30 02:30:51,027 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-output0 based on ^bcm2835-isp-\w+$
2021-04-30 02:30:51,029 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-encode based on ^bcm2835-codec-\w+$
2021-04-30 02:30:51,067 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-stats2 based on ^bcm2835-isp-\w+$
...
2021-04-30 02:58:00,215 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture0 based on ^bcm2835-isp-\w+$
2021-04-30 02:58:00,216 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-isp based on ^bcm2835-codec-\w+$
2021-04-30 02:58:00,249 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-decode based on ^bcm2835-codec-\w+$
2021-04-30 02:58:00,252 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture1 based on ^bcm2835-isp-\w+$
2021-04-30 02:58:00,254 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-output0 based on ^bcm2835-isp-\w+$
2021-04-30 02:58:00,258 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-encode based on ^bcm2835-codec-\w+$
2021-04-30 02:58:00,290 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-stats2 based on ^bcm2835-isp-\w+$

From that I'd say you opened the settings window 4 times.
The above devices are normal rpi related things that aren't actually cameras, so they get excluded.

Are you certain it's not being show in the settings window under plugins?
[image]

Navyguy330, 2021-04-30:
damn i am an idiot, I thought it was gonna be in the bar at the top with some of the others.  good lord.  thanks. lol still bummer I can't actually use it hahah. you should work with sd to see if you can get it working with it. cus that is such a valuable resource as well as this one, and to only be able to have one or the other lol.  

btw,  can you disable detective,  make the settings you want,  save them THEN reactivate detective?

Navyguy330, 2021-04-30:
I will say messing with the settings, the auto focus button never lets me uncheck it.  not sure if related to detective etc. the preview DOES update other settings tho.  not sure if it would should or could with detective

Navyguy330, 2021-04-30:
thanks for the help though, and keep up the good work. this is something I was looking for for a while.

The-EG, 2021-04-30:
> I will say messing with the settings, the auto focus button never lets me uncheck it. not sure if related to detective etc. the preview DOES update other settings tho. not sure if it would should or could with detective

Sounds like #12 . There are few settings with a few cameras that are doing that; still trying to understand why. If the preview is updating then it sounds like you either don't have TSD premium active or it's not currently running...

> thanks for the help though, and keep up the good work. this is something I was looking for for a while.

Thanks!

closed 2021-04-30 by The-EG

Navyguy330, 2021-04-30:
yeah, nothing is printing right now, I do pay for premium.  just not using anything so it must not be active unless I print.  interesting that is odd. 

well if you need a Guinee pig for the c920s let me know.   (esp someone with detective)

thanks again.