The-EG/OctoPrint-CameraSettings #15: [FR] Implement Controls for [Camera Name]

AlAmantea, 2021-04-30:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "hue_auto": {
    "type": "bool",
    "default": "0",
    "value": "0"
  }
}

The-EG, 2021-04-30:
This is now available in pre-release 0.1.1rc1.

You can get it now by changing to the Release Candidate channel for this plugin in the Software Update settings:
[image]

After some testing, this will be released into an official release (0.1.1)

AlAmantea, 2021-04-30:
Thanks Al Amantea  From: Taylor TalkingtonSent: Friday, April 30, 2021 8:18 AMTo: The-EG/OctoPrint-CameraSettingsCc: AlAmantea; AuthorSubject: Re: [The-EG/OctoPrint-CameraSettings] [FR] Implement Controls for [Camera Name] (#15) This is now available in pre-release 0.1.1rc1.You can get it now by changing to the Release Candidate channel for this plugin in the Software Update settings:After some testing, this will be released into an official release (0.1.1)

commit bb5c30b8 (2021-04-30): Add 'Hue Auto' control for unspecified camera (fix #15)
closed 2021-05-01 by The-EG