The-EG/OctoPrint-CameraSettings #18: [BUG] Some Settings do not work

LordIkol, 2021-04-30:
**Observed behavior**
Some of the Sliders and Checkboxes can not be changed. 
Namely "Focus Auto" Checkbox can not be unticked
Focus Absolute slider stuck at 30 
Zoom Absolute stuck at 100 
Pan & tilt don't work either but this could be camera issue since i never got them to work with uvcdynctrl either. but the Focus and Zoom should work.

**Expected behavior**
I should be able to change them.

**OctoPrint Version**
1.6.0

**CameraSettings Plugin Version**
0.1.0

**Camera**
Logitec StreamCam

**Linux/OctoPi**
0.18.0

**Other Plugins**
Octolapse is the only other plugin that uses the cam stream

The-EG, 2021-04-30:
Please open the settings window, go to Logging and add a debug entry for this plugin. It should look like this:
[image]

Then try to use those settings again, go back to that same screen and click the download button next to 'octoprint.log'

Unfortunately, there have been a few controls for a few cameras that haven't been responsive. I'm not sure if there's much I can do about it, but this will give me an idea if there's something wrong with what's being sent back and forth.

The-EG, 2021-05-01:
My fix for #12 may work here too.

Please see this comment on how to test the fix: https://github.com/The-EG/OctoPrint-CameraSettings/issues/12#issuecomment-830636480

The-EG, 2021-05-01:
For now I'm going to say this is a duplicate of #12 

Please install the latest rc, 0.1.2rc1. You will need to select the 'Release Candidate' channel for this plugin in OctoPrints Software Update settings if you haven't already.

If this issue persists with in the last rc, this issue can be reopened.

closed 2021-05-01 by The-EG