The-EG/OctoPrint-CameraSettings #19: [FR] Implement Controls for [Camera Name]

Oxize, 2021-04-30:
**Camera manufacturer and model (ie. NexiGO N660)**: Vitade HD960

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:

{
  "zoom_continuous": {
    "type": "int",
    "min": "0",
    "max": "5",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

Oxize, 2021-04-30:
What means the FR tag in the title. Just curious?

The-EG, 2021-04-30:
> What means the FR tag in the title. Just curious?
Feature Request

Oxize, 2021-04-30:
Ah cool. I am so happy this Octoprint is working out so cool with my Bresser Saurus. Bresser told, nha OctoPi doesnt work with our printers. Keep the stuff with Polarcloud. So i saw in the firmware, that the Saurus was party FlashForge. Used the 3D printer Plugin for that, and voila it was connecting after a small nano entry in the rules.

The-EG, 2021-04-30:
Good to hear, I'm sure the OctoPrint devs would be happy to hear that (I just work on this plugin and a few other things)!

I've added 'Zoom Continuous' to the RC branch...there will be a release candidate soon.

Oxize, 2021-04-30:
thanks, looking forward to it. Thanks for your hard work. Stay Save!

The-EG, 2021-04-30:
This is now available in pre-release 0.1.1rc1.

You can get it now by changing to the Release Candidate channel for this plugin in the Software Update settings:
[image]

After some testing, this will be released into an official release (0.1.1)

commit aef638bb (2021-04-30): Add Zoom Continuous control for Vitade HD960 (fix #19)
closed 2021-05-01 by The-EG