The-EG/OctoPrint-CameraSettings #22: [FR] Implement Controls for [Logitech C922 Pro Stream]

Quintin9801, 2021-04-30:
**Camera manufacturer and model (ie. NexiGO N660)**:
Logitech C922 Pro Stream
**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "led1_mode": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Off"
      },
      {
        "value": "1",
        "desc": "On"
      },
      {
        "value": "2",
        "desc": "Blink"
      },
      {
        "value": "3",
        "desc": "Auto"
      }
    ],
    "min": "0",
    "max": "3",
    "default": "0",
    "value": "3"
  },
  "led1_frequency": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "1",
    "default": "0",
    "value": "24"
  }
}

The-EG, 2021-04-30:
Oops, I missed the ref for this one. This is in commit 9b8307b589858dfaf70d301da1b46007e80846c9

It'll be in an RC release soon.

The-EG, 2021-04-30:
This is now available in pre-release 0.1.1rc1.

You can get it now by changing to the Release Candidate channel for this plugin in the Software Update settings:
[image]

After some testing, this will be released into an official release (0.1.1)

commit 222b8762 (2021-04-30): Add LED Mode and LED Frequency controls

Requested for multiple cameras:
 - Logitech C270 HD (fix #16)
 - Logitech StreamCam (fix #17)
 - Logitech C922 Pro Stream (fix #22)
closed 2021-05-01 by The-EG