The-EG/OctoPrint-CameraSettings #24: [FR] Raspicam - change resolution without a reboot?

Naugrimohtar, 2021-04-30:
So today I can change RaspiCam resolution via editing '/boot/octopi.txt' an then rebooting...
Is there a way to change it on the fly? Incorporate into this plugin? This would be hugely helpful!

The-EG, 2021-04-30:
I'll put this on the TODO list to research. 

It's technically possible to send a request to set the resolution in a similar manner to how the controls are set, but since mjpg-streamer is expecting the stream to be of a particular size I'm not sure how well it would behave if it suddenly changes without warning.

uStreamer might work better since it seems to be able to handle changes and even a lack of the camera stream better...

In any case, this will require some testing and thought!

BobCraig10, 2021-04-30:
That would be a great addition to a WONDERFUL plugin!  Great job!!

The-EG, 2021-04-30:
If you ever want to use higher resolutions (at the cost of FPS), check out the modprobe options described: [here](https://github.com/pikvm/ustreamer#raspberry-pi-camera-example). (uStreamer itself is also a nice alternative to mjpg-streamer)