The-EG/OctoPrint-CameraSettings #26: [FR] Implement Controls for Logitech C905 - most importantly, manual focus!!

diaconesq, 2021-04-30:
Logitech C905

{
  "focus": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "1",
    "default": "254",
    "value": "113"
  },
  "led1_mode": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Off"
      },
      {
        "value": "1",
        "desc": "On"
      },
      {
        "value": "2",
        "desc": "Blinking"
      },
      {
        "value": "3",
        "desc": "Auto"
      }
    ],
    "min": "0",
    "max": "3",
    "default": "3",
    "value": "3"
  },
  "led1_frequency": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "disable_video_processing": {
    "type": "bool",
    "default": "0",
    "value": "0"
  },
  "raw_bits_per_pixel": {
    "type": "int",
    "min": "0",
    "max": "1",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

diaconesq, 2021-04-30:
Low values are for far focus, high for close-up.

The-EG, 2021-04-30:
LED Mode and Frequency are already implemented in 0.1.1rc1 (from a few other users with various Logitech cameras).

I'll add the other shortly and get 0.1.1rc2 up.

The-EG, 2021-04-30:
This is now available in pre-release 0.1.1rc2.

You can get it now by changing to the Release Candidate channel for this plugin in the Software Update settings:
[image]

After some testing, this will be released into an official release (0.1.1)

commit d800c4e5 (2021-04-30): Implement controls for Logitech C905 (fix #26)

 - Focus
 - Disable Video Processing
 - Raw Bits Per Pixel
closed 2021-05-01 by The-EG