The-EG/OctoPrint-CameraSettings #28: [BUG] Preview does not reflect selected camera (multiple camera's installed)

ruedli, 2021-05-01:
The stream and webcam plugins sohw the different camera's when selected (I have two camera's installed and plugin multicam configured).

When in OctoPrint-Camerasettings I select the second camera, this selection is reflected in all the settings (they differ). The preview however STAYS on the first camera. Even enabling and disabling preview does not "switch" camera's. 

**Expected behavior**
When the pulldown is operated, the preview should reflect the selected camera.

**OctoPrint Version**
1.6.0
**CameraSettings Plugin Version**
0.1.1

**Camera**
Logitech C920
USB Camera (noname/nobrand known)
[image]

[image]

**Other Plugins**
Octolapse

The-EG, 2021-05-01:
Indeed, the plugin currently has no way of knowing what camera you have setup in OctoPrint, it just uses whatever is in the settings. 

It would be possible to allow a custom preview URL to be setup for multiple camera setups like this, but since it's the exception and **not** the standard use case, I need to make sure that it doesn't effect how most users will use it. Ie. I will not require all users to specify a URL when they have a single camera.

The-EG, 2021-05-01:
[image]

I reference the wrong issue here! That commit is for #12...disregard.

ruedli, 2021-05-01:
I do not know whether this is of help for you, but here is some anyhow ;-) . Two other plugins seem to pick up the camera based info upon a pull-down: "steams" and "webcam" e.g. in their side panels. I have setup the second camera in the "multicam" plugin and the other plugins probably got info from there:

[image]

ruedli, 2021-05-01:
Let me know if you want me to pull some information how these settings are stored (and how to pull it, I am not a python programmer :-)  ). I feel this is quite a generalized way of configuring things, probably just a second block of info. Note that e.g. Printoid automatically shows two streams remotely, so I must be in the config info somehow.

The-EG, 2021-05-01:
This is good info. I imagine that most users who have this setup will probably have the 'multicam' plugin so I'll see if I can use any info from that. Thanks!

ruedli, 2021-05-01:
OK, I call octopi/API/setting, got the looooooong json and found under plugins:multicam this:

[image]

You can probably install multicam, add a second profile pointing to your same camera, temporarily "ruin" the setting of the octoprint webcam (so the default profile) and debug it. Let me know if you need more info...

ruedli, 2021-05-01:
Out of intesets: you allrerady have a pulldown, it lists
- HD Pro Webcam C920
- USB 2.0 PC Camera:USB 2.0 PC Cam

Where are you getting those two from? They are not anywhere in the settings. Is that from enumeration of the video steams?

The-EG, 2021-05-01:
> Is that from enumeration of the video streams?

Close, I'm poking through the video-4-linux sysfs info and pulling in the names for all of the available video devices. Anything that doesn't report any control capabilities is automatically filtered out, and some of the raspberry pi GPU stuff is manually filtered out as well.

What's left is what should be the usable cameras on the system (I end up with pairs of names and devices, ie. /dev/video0)

By design, this is done completely agnostically from the streamer (mjpg-streamer). This is because I want to be compatible with *any* streamer; in fact I use uStreamer in my environments instead of mjpg-streamer.

Because of that, I can't rely on mjpg-streamer even being present and that means that URLs, etc. are completely unpredictable.

As-is, this will definitely require some manual configuration by the user. Even if I can get the stream URLs from the multicam plugin, I'll have no way of automatically mapping which camera is which. But I think that's a fairly minimal amount of configuration and acceptable.

ruedli, 2021-05-01:
Yes, minimal is OK. For sure the first profile is the default cam the stream. As you can see, it is the mpeg-streamer, and I had to make sure it always maps to the right stream by filtering on USB info (serial number, manufacturer, that it is a streaming url etc.) I added mine to multicam in the order of /dev/videoX . The streams are alway pointing to the same camera. Hopyfully you can also poke the info in that order, then the order of "multicam" correspondent to your order if I am not wrong. Alternatively, you can pull the names from multicam and have the user "choose"  from a pulldown which camera of yours applies. But as mentioned, I believe the order is the same. good stuff! I like your plugin. It is the first not to depend on the streamer choice of Octoprint. Are there advantages to using uStreamer? How would you need to setup the deamon processes for that? I "just" copied the existing deamon config and added a new deamon...

The-EG, 2021-05-01:
> Are there advantages to using uStreamer? How would you need to setup the deamon processes for that?

I haven't done side by side or scientific comparison but uStreamer's performance is quite good and feels more stable.

 I have 2 instances of OP, each with a camera on the same rpi and even streaming both at the same time, neither instance of uStreamer is over 8% CPU. That's pretty impressive in my opinion since my raspicam 1.3 is running at 2592x1944 at 10fps.

It also behaves much better when something goes wrong with the camera, you'll get an image that says 'no stream available' instead of just not working and you can configure it to retry, etc...

Setup is very similar to mjpg-streamer, although the stream URL is a tiny bit different. I just created my own systemd services.

https://github.com/pikvm/ustreamer

ruedli, 2021-05-01:
Yes, that is good performance. I am running similar resolution / speeds, but my problems went away by throwing in an Rpi 4 . Also it went a leaptimes better when I changed to a SSD setup versus SD. Since then, I never worry about performance and I can see the two full size video streams, upload files AND arc-weld / process them without any noticeable "freezes" of extruder movement. It also boots much faster.

I'll look into uStreamer as well. Interesting.

At first I thought, ah, I already have control over the camera, through mentioned Octolapse. But your plugin takes it way further, supporting many more camera's, even my "chinese noname" extruder boroscope! 

No I learned that is is streamer unagnostic. Well done!

The-EG, 2021-05-15:
This is now available in 0.2.0rc1.

The-EG, 2021-05-15:
After an oops that is now fixed: now available in 0.2.0rc1.

ruedli, 2021-05-16:
@The-EG The Well Taylor, I tried 0.2.0rc1. Enabled the multicam integration, filled in my multicam names to the camera names you detected and.... It works splendidly! Well done, you're a champ!

As far as I am concerned this is a GO for the rc.

[image]

[image]

PR #35 (2021-05-15): Support multiple camera previews with OctoPrint-MultiCam
Initial multiple camera preview support.

This will only work if you also have the most recent version of the 'MultiCam' plugin installed.

With these changes, the plugin will detect that MultiCam is installed and allow mapping detected cameras to profiles in MultiCam:
[image]

[image]

And, if 'OctoPrint-MultiCam Integration' is enabled, the preview will use the URL and flip, rotate settings from the MultiCam profile:

https://user-images.githubusercontent.com/7342077/118275487-77066780-b494-11eb-981f-000ff0dd9ac4.mp4

If anyone would l [... cut]
commit 4c257b3e (2021-05-15): Support multiple camera preview with OctoPrint-MultiCam (fix #28)
commit 8392f193 (2021-05-15): Support multiple camera preview with OctoPrint-MultiCam (fix #28)
closed 2021-05-16 by ruedli