The-EG/OctoPrint-CameraSettings #29: [FR] New Feature Request

KenLucke, 2021-05-01:
**Description of Desired Feature**
"Reset camera to default" button.

I've messed with some controls with the plugin, but there's no way I've found to just simply reset everything back to defaults.  It would be nice to have a one-click way to reset all the changes made.

The-EG, 2021-05-01:
That's a reasonable request and the camera does report default settings for each control. Targeting version 0.2.0

The-EG, 2021-05-24:
This is now in 0.2.0rc3. You can try it out now by selecting the "Release Candidate" channel for this plugin on the Software Update screen of the OctoPrint Settings window.

closed 2021-05-29 by The-EG

KenLucke, 2021-05-29:
Sorry, I didn't see the notice of rc3 a few days ago, just saw the thread closure now.  Tested minimally, and seems to function.

commit bc7ebc46 (2021-05-24): Add 'Restore Defaults' Button (fix #29)