The-EG/OctoPrint-CameraSettings #31: [FR] New Feature Request

Peuqui, 2021-05-02:
Great plugin btw!!! Until now, I used the Octolapse plugin only for adjusting my camera settings and not for timelapses :-)
My suggestion:
Let us modify Camera settings "on the fly": Let the settings window appear when a button on the main screen is pressed and/or better: let it run in a tab on the right side of the window.
It is a bit fiddly to go into the settings menu, choose your plugin and then adjust the settings...
Keep up your good work!

Best regards,
Peuqui

The-EG, 2021-05-02:
Hi there, thank you and thank you for trying it out.

In regards to the placement, it actually started out on its own tab and was later moved to the settings window after some discussion: 
https://community.octoprint.org/t/camera-settings-plugin-seeking-feedback-and-testing/33139

It's location in the settings is deliberate.

Peuqui, 2021-05-02:
Hi and thanks for your fast reply!
Sure, for most users, the position in the settings may be suitable.
Unfortunately, I need to reconfigure my camera settings fairly often to reflect the situation of lighting. E.g. lighting changes in regard of the size and color the 3D print and the ambient light brightness.
Suggestion:
You can add a button in the settings window, where everybody can choose, if one wants to add this really nice feature to a tab or not.

The-EG, 2021-05-03:
I'm not completely opposed to idea, but it's also not going to be be prioritized over things that add new functionality.

I'll do this: I'll leave it open as a reminder it to at least revisit it at some point.