The-EG/OctoPrint-CameraSettings #32: [BUG] Settings panel is mostly blank, no dropdowns

bruno615, 2021-05-03:
**Observed behavior**
What is happening or not working?

I can only see Show Preview (and the preview).  The dropdown below it is unselectable and has no options.  There's then a huge gap in blank space before save settings (which I assume looks about right)

**Expected behavior**
What should be happening?

I should have the ability to select the camera, any related settings, etc.

**OctoPrint Version**
What version of OctoPrint are you using?

**CameraSettings Plugin Version**
What version of this plugin are you using?

**Camera**
What camera are you using?
Raspi Cam.  I did the steps in the setup guide specific to the Raspi Cam too (with reboot).  
[image]

**Linux/OctoPi**
What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
Octopi 1.6.0

**Other Plugins**
Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
Dont think so.

The-EG, 2021-05-03:
Hey there, if you close the settings window and then re-open it, does this still happen?

bruno615, 2021-05-03:
I disabled all theming plugins and rebooted.  This solved the issue.  I can go back and figure out which one was responsible.

closed 2021-05-03 by bruno615

The-EG, 2021-05-03:
> I disabled all theming plugins and rebooted. This solved the issue. I can go back and figure out which one was responsible.

Ok. I've tested this with UI Customizer and it *should* be working so I definitely want to know if something is throwing it off.

I suspect what has happened is that you opened up the settings window a bit before the web client was fully connected. In that case, the information that gets sent can be 'lost'. Closing the settings window and opening it back up will usually clear it up.