The-EG/OctoPrint-CameraSettings #33: [BUG] Working Only with Genuine RPi Cam ?

arut16, 2021-05-04:
**Observed behavior**
Video stream stop after RPi cam setup for the plugin : black screen.

**OctoPrint Version**
1.6.0

**CameraSettings Plugin Version**
0.1.2

**Camera**
Chinese copy from AliExpress : [Here](https://fr.aliexpress.com/item/32881466491.html?spm=a2g0s.9042311.0.0.27426c37IC5U3t)

**Linux/OctoPi**
0.17.0

**Other Plugins**
Octolapse (installed but disabled in octolapse settings)

**Does this plugin working only with genuine RPi Cameras (v1 and v2) ?**

The-EG, 2021-05-04:
I haven't tested a generic version personally, but I believe a couple users reported using 'ArduCams' and one similar to yours without issue. 

When you say 'black screen' do you mean completely black with no text or the 'Couldn't load stream' screen? What happens if you look at the stream directly? (http://octopi.local/webcam/?action=stream)

arut16, 2021-05-04:
Yes. A complete black screen without any text on it. And same black screen with direct stream.

The-EG, 2021-05-04:
Have you tried changing any of the settings? Is it possible that the exposure is set to something where it's not getting enough light, etc?

arut16, 2021-05-04:
I've tried some settings but not sure about exposure. Will check SASP !

closed 2021-05-07 by The-EG