The-EG/OctoPrint-CameraSettings #34: [BUG] White balance zeros out. But is still applied correctly.

CyanicTempest, 2021-05-12:
**Observed behavior**
What is happening or not working?
When setting the white balance the camera updates correctly but the UI defaults to 0. Which is then reflected in the saved preset.

**Expected behavior**
What should be happening?
I'd expect the white balance to reflect the current setting of the camera. (This is probably a quirk of the cheap camera.)

**OctoPrint Version**
What version of OctoPrint are you using?
1.6.0

**CameraSettings Plugin Version**
What version of this plugin are you using?

**Camera**
What camera are you using?
A budget generic webcam. "Webcam gadget HS: V380 UVC Camera"

**Linux/OctoPi**
What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
Version 0.18.0, running on Raspberry Pi 3 Model B Rev 1.2

**Other Plugins**
Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
No.

The-EG, 2021-05-12:
Hi there, thanks for trying the plugin out, the report, and actually filling out the template!

That's odd that the setting works but the slider is set back to 0.

A debug log would probably helpful.
1. Open up the OctoPrint settings window
2. Go to 'Logging' under OctoPrint settings
3. Under 'Logging Levels' select this plugin (`octoprint.plugins.camerasettings`) and leave the level at `DEBUG`.

It should look like:
[image]

After that, go back to the camera settings and play with the controls, especially white balance again. Then come back to the 'Logging' page, click the download button for 'octoprint.log' and attach it here.  You can then delete the debug line for the plugin if you want.

The-EG, 2021-05-19:
Feel free to reopen with debug logs.

closed 2021-05-19 by The-EG