The-EG/OctoPrint-CameraSettings #37: [FR] Implement Controls for [Camera Name]

rflegel, 2021-05-16:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "white_balance_red_component": {
    "type": "int",
    "min": "1",
    "max": "500",
    "step": "1",
    "default": "100",
    "value": "100",
    "flags": "inactive"
  },
  "white_balance_blue_component": {
    "type": "int",
    "min": "1",
    "max": "500",
    "step": "1",
    "default": "100",
    "value": "100",
    "flags": "inactive"
  },
  "iris_absolute": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "0",
    "default": "0",
    "value": "0"
  },
  "iris_relative": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  }
}

The-EG, 2021-05-16:
Hi there, what kind of camera is this?

Also, the iris controls look a bit odd. I'll add these, but I'm not sure if the 'iris' ones will have any use since the only valid value is '0.'

The-EG, 2021-05-17:
Now in 0.2.0rc2

rflegel, 2021-05-17:
This is the Amazon Link to the camera. Will that work for you?

https://www.amazon.com/gp/product/B08RHTG845/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

Thanks

The-EG, 2021-05-17:
> This is the Amazon Link to the camera. Will that work for you?

Perfect, thank you. It's not critical, but it helps me understand what types of cameras have been used, especially if other users have similar cameras.

commit 6b63f4cc (2021-05-17): Add additional camera controls

 - white_balance_red_component, white_balance_blue_component, iris_absolute,
    iris_relative: fix #37
 - white_balance_component_auto: fix #36
closed 2021-05-29 by The-EG