The-EG/OctoPrint-CameraSettings #38: [FR] Implement Controls for Zooming the camera

Vantiveman, 2021-05-19:
**Camera manufacturer and model (ie. NexiGO N660)**:
Raspberry Pi Camera
**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
This plug-in is great, with more bells and whistles than I can swing a cat at. However, the one thing missing that would be great is a zoom control.
Any chance of adding a zoom? While many camera's (mine included) don't have a zoomable lens, a simple image zoom would be nice even though the result would be a lower resolution.

The-EG, 2021-05-19:
Hello, thanks for using the plugin and for the FR.

Unfortunately, that is something that's probably never going to be in scope for this plugin. As you've noticed, this plugin can only control what actual capabilities the camera has and reports. 

This plugin is designed to function as independently as possible from the streamer (mjpg-streamer, ustreamer, etc.) and other OctoPrint plugins being used as possible. That allows it to be useful for the most use cases as possible while being less complex.

Because of that, the plugin is really only aware that there are cameras, that each one of them have certain capabilities and the parameters of those capabilities. This means that the plugin doesn't have access to the stream itself or any information about it (resolution, etc.).

You may be saying, 'Well, how do you show the preview then!?' The preview is just an image (as far as the plugin is concerned), whose source address is set from OctoPrint's camera settings. If you change the stream URL in OctoPrint, the preview changes too, etc.

This sort of functionality could be implemented as a separate plugin that alters the stream image, but that would get tricky since it would need to do so on every instance of it and there are multiple plugins that can show the stream in multiple places.

closed 2021-05-19 by The-EG

Vantiveman, 2021-05-19:
Hi,

Thank you for your reply. I hadn’t realised that the plugin didn’t have access to resolution data so I understand the difficulty (impossibility) of my request.

Thanks for your response. It remains a great plugin and even as it stands does far more than I thought possible. Watching a print in embossed mode is strangely fascinating :)

Al