The-EG/OctoPrint-CameraSettings #39: [FR] Implement Controls for [Camera Name]

JORATORCA, 2021-05-23:
**Camera manufacturer and model (ie. NexiGO N660)**: PHILIPS SPC 900NC WEBCAM

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "band_stop_filter": {
    "type": "bool",
    "default": "1",
    "value": "1"
  },
  "auto_contour": {
    "type": "bool",
    "default": "0",
    "value": "0",
    "flags": "update"
  },
  "contour": {
    "type": "int",
    "min": "0",
    "max": "63",
    "step": "1",
    "default": "16",
    "value": "16",
    "flags": "slider"
  },
  "dynamic_noise_reduction": {
    "type": "int",
    "min": "0",
    "max": "3",
    "step": "1",
    "default": "1",
    "value": "1"
  },
  "auto_white_balance_speed": {
    "type": "int",
    "min": "1",
    "max": "32",
    "step": "1",
    "default": "1",
    "value": "1"
  },
  "auto_white_balance_delay": {
    "type": "int",
    "min": "0",
    "max": "63",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "save_user_settings": {
    "type": "button",
    "flags": "write"
  },
  "restore_user_settings": {
    "type": "button",
    "flags": "update"
  },
  "restore_factory_settings": {
    "type": "button",
    "flags": "update"
  }
}

The-EG, 2021-05-23:
> "save_user_settings": {
> "type": "button",
> "flags": "write"
> },
> "restore_user_settings": {
> "type": "button",
> "flags": "update"
> },
> "restore_factory_settings": {
> "type": "button",
> "flags": "update"
> }

Well that's a new one. Normally I can add controls very easily, but this is the first time I've seen 'buttons.' I'll have to research these a bit.

The-EG, 2021-05-24:
With the exception of the last 3 'buttons,' this is now in 0.2.0rc3.

commit 89846702 (2021-05-24): Add camera controls for Philips SPC 900NC (fix #39)

Added controls:
 - band_stop_filter
 - auto_contour
 - contour
 - dynamic_noise_reduction
 - auto_white_balance_speed
 - auto_white_balance_delay
closed 2021-05-29 by The-EG