The-EG/OctoPrint-CameraSettings #40: [FR] Implement Controls for [Miceshu/Jieli/USB PHY 2.0]

krudoy, 2021-05-24:
**Camera manufacturer and model (Miceshu/Jieli/USB PHY 2.0)**:
    ATTRS{product}=="USB PHY 2.0"
    ATTRS{idVendor}=="1224"
    ATTRS{manufacturer}=="Jieli Technology"
    ATTRS{bConfigurationValue}=="1"
    ATTRS{idProduct}=="2825"

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "white_balance_red_component": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "0",
    "default": "128",
    "value": "128"
  },
  "white_balance_blue_component": {
    "type": "int",
    "min": "0",
    "max": "255",
    "step": "1",
    "default": "128",
    "value": "128"
  },
  "iris_absolute": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "0",
    "default": "0",
    "value": "0"
  },
  "iris_relative": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  }
}

The-EG, 2021-05-24:
Hey there, these should already be implemented in the current release candidate. You can try it out by selecting the 'Release Candidate' channel for this plugin in the Software Update page in OctoPrint's settings. That should prompt you to update to 0.2.0rc2.

The-EG, 2021-05-24:
same controls as #37

closed 2021-05-29 by The-EG