The-EG/OctoPrint-CameraSettings #45: [BUG] Cannot install in Octoprint 1.6.1

DAveShillito, 2021-06-04:
**Observed behavior**
Cannot install in Octoprint 1.6.1

As can be seen below the Octoprint UI says it is incompatible
[image]

I tried to manually install from https://github.com/The-EG/OctoPrint-CameraSettings/archive/main.zip
[image]

This gave the error as below.
[image]

The error text is

> Installing plugin from https://github.com/The-EG/OctoPrint-CameraSettings/archive/main.zip.../home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmp_BP4Ga/main.zip --no-cache-dirLooking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> Processing /tmp/tmp_BP4Ga/main.zip
> ERROR: Package 'OctoPrint-CameraSettings' requires a different Python: 2.7.16 not in '>=3, <4'
> Error!
> Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

As instructed I looked at "plugin_pluginmanager_console.log" and this contains

> 2021-06-03 15:26:59,124 > Cleaning up plugin homeassistant...
> 2021-06-03 16:50:49,931   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpFYwyeA/main.zip --no-cache-dir
> 2021-06-03 16:51:16,380 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> 2021-06-03 16:51:16,383 > Processing /tmp/tmpFYwyeA/main.zip
> 2021-06-03 16:51:42,271 ! ERROR: Package 'OctoPrint-CameraSettings' requires a different Python: 2.7.16 not in '>=3, <4'
> 2021-06-03 20:52:00,761   /home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmp_BP4Ga/main.zip --no-cache-dir
> 2021-06-03 20:52:23,706 > Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
> 2021-06-03 20:52:23,730 > Processing /tmp/tmp_BP4Ga/main.zip
> 2021-06-03 20:52:48,522 ! ERROR: Package 'OctoPrint-CameraSettings' requires a different Python: 2.7.16 not in '>=3, <4'

(The "Cleaning up plugin homeassistant" is the operation I performed before trying to install OctoPrint-CameraSettings, I only include it for completeness)

**Expected behavior**
I should be able to install

**OctoPrint Version**
1.6.1

**CameraSettings Plugin Version**
I am trying to install 0.2.0 (2021-05-29)

**Camera**
Microsoft Lifecam HD6000

**Linux/OctoPi**
OctoPi Version 0.17.0, running on Raspberry Pi Model B Rev 2

**Other Plugins**
Installed 3rd party plugins are 

Bed Leveling Plugin (2.1.2)
Custom Control Editor (0.2.2)
DisplayLayerProgress Plugin (1.20.0)
Filament Manager (1.8.0)
FileManager (0.1.6)
HomeAssistant Discovery (3.3.0)
Preheat Button (0.8.0)
PrintJobHistory (1.14.0)
SpoolManager Plugin (1.4.3)

The-EG, 2021-06-04:
Hi there, as the error message indicates, this plugin is not compatible with Python 2. You will need to upgrade your OctoPrint install to Python 3 to use it.

Please see: 
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

closed 2021-06-04 by The-EG

DAveShillito, 2021-06-04:
Adding an additional note in case people find this (not) bug in the future.
This is an example of the OctoPrint (program) being different to the OctoPi (distribution)
I had been upgrading OctoPrint, but had missed the fact that the version of OctoPi I was using did not include Python3.

As well as the link above, this one describes it well.
https://octoprint.org/blog/2021/01/25/octopi-release-0-18-0/

Thanks @The-EG for the prompt reply, I figured it had to be something I was missing since other 190 people had successfully installed this week :)

The-EG, 2021-06-04:
Actually, it's possible to upgrade your existing OctoPi image to Python 3 in most cases. That is what the post I linked explains and it also contains a script that automates the process.

OctoPi is actually just a configured version of Raspberry Pi Linux with OctoPrint, mjpg-streamer and other things pre-configured. Version 0.17 certainly does have Python 3 available, it just has to be installed and enabled.

There's certainly nothing wrong with just flashing 0.18 as a means to upgrade, but if you want to upgrade your existing setup just run the script as described in the blog post.

DAveShillito, 2021-06-05:
Thanks @The-EG 

I had followed the script in the link you suggested, just because I wanted to print something and the quality of my image had been annoying me and I wanted to use your plugin to fix it  - it worked great BTW - Thanks :)

I will eventually reflash since as it says in the link I posted.

> Normally there’d be no pressing need to reflash, however since this image finally ditches Python 2 in favour of Python 3, which apart from continued support also adds some nice performance improvements, you might want to switch sooner rather than later.

As I mentioned I was mainly updating this issue to clarify for others reading it in the future why although I had thought I was keeping up to date I had forgotten what was provided by OctoPrint, and what was provided by OctoPi (although I suppose this will only be an issue for this brief transitional time)

Once again, thanks for the plugin and your support. :)