The-EG/OctoPrint-CameraSettings #46: [FR] Implement Controls for Logitech Quickcam Orbit Sphere AF

perkinbr, 2021-06-23:
Logitech Quickcam Orbit Sphere AF
Maybe you don't want to support these, but I thought I'd file the issue anyway.

{
  "pan_relative": {
    "type": "int",
    "min": "-4480",
    "max": "4480",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  },
  "tilt_relative": {
    "type": "int",
    "min": "-1920",
    "max": "1920",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  },
  "pan_reset": {
    "type": "button",
    "flags": "write"
  },
  "tilt_reset": {
    "type": "button",
    "flags": "write"
  }
}

The-EG, 2021-06-23:
Sure.

I haven't figured out a good way to support 'button' controls yet, but I can add the others easily enough.

The-EG, 2021-06-24:
If you can, please test this out by installing the plugin from this URL:
https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fixes/more_controls.zip

That should implement Pan and Tilt and git rid of the warning.

perkinbr, 2021-06-28:
I tried it.  I think maybe that's not really the right thing to do.
Since it's a relative tlit and pan setting, it also needs a button type solution since it moves every time you set the setting.
Sorry, I guess I should have explained what that setting did.

The-EG, 2021-06-28:
Interesting, ok. This is a case of the camera manufacturers reporting things a bit wrong then. I'll take a look, thanks for the feedback!

The-EG, 2021-07-01:
Thinking about this some more...so I take it that moving the control to the 'left' (decrease, negative value), decreases the value by that amount and then resets the value to 0 each time? An increase/right would have the opposite effect?

perkinbr, 2021-07-01:
Yes, that's how it works.

philsynek, 2022-03-16:
Hey,
I also have a Logitech QuickCam Orbit/Sphere and would really like to have pan/tilt working. Is there a way I can support here?
Thanks!
Phil

The-EG, 2022-03-18:
Hi @philsynek , 
I don't have a camera with these capabilities, so I'll really need users who are willing to test and provide feedback, and of course the time to actually work on it. 
If you are willing to be a guinea pig and test things out, that'd be helpful. I'll see if I can put together a dev branch to try out this weekend.

philsynek, 2022-03-18:
That's why I am asking. Let me be your guinea pig! ;) But I will need some instructions what to do exactly, since I am pretty new to OctoPrint. Take your time!

perkinbr, 2022-03-18:
I don't use octoprint anymore but do you see the pan settings in v4l-ctrl?

philsynek, 2022-03-19:
I think so:

pi@octopi:~ $ v4l2-ctl -L
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=32 value=32
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=32 value=32
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=158
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=0 max=10000 step=10 default=4000 value=10000 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=224 value=224
         backlight_compensation 0x0098091c (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=1 max=10000 step=1 default=166 value=155 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
                   **pan_relative 0x009a0904 (int)    : min=-4480 max=4480 step=0 default=0 value=0 flags=write-only
                  tilt_relative 0x009a0905 (int)    : min=-1920 max=1920 step=0 default=0 value=0 flags=write-only
                      pan_reset 0x009a0906 (button) : flags=write-only
                     tilt_reset 0x009a0907 (button) : flags=write-only**
                          focus 0x009a090a (int)    : min=0 max=255 step=1 default=0 value=0
                      led1_mode 0x0a046d05 (menu)   : min=0 max=3 default=3 value=3
                                0: Off
                                1: On
                                2: Blinking
                                3: Auto
                 led1_frequency 0x0a046d06 (int)    : min=0 max=255 step=1 default=0 value=0
       disable_video_processing 0x0a046d71 (bool)   : default=0 value=0
             raw_bits_per_pixel 0x0a046d72 (int)    : min=0 max=1 step=1 default=0 value=0

perkinbr, 2022-03-19:
Great. Every new PI I've installed on needs this:

https://joshcurry.co.uk/posts/pan-tilt-controls-not-available-on-logitech-orbit-sphere-af-on-linux
/

But maybe you have that already

The-EG, 2022-03-19:
It's not a matter of the controls being available, I just have to figure out a way to add them to the UI that makes sense :smile: 

The issue at hand is that these are relative controls and so a slider isn't really the right way to represent them. And the 'reset' ones will need to be buttons, but I'm not entirely sure how those actually work and without a camera to test with, that means I'll have to just guess and have some users test it out for me, provide feedback, fix and repeat.

philsynek, 2022-03-19:
> Great. Every new PI I've installed on needs this: https://joshcurry.co.uk/posts/pan-tilt-controls-not-available-on-logitech-orbit-sphere-af-on-linux / But maybe you have that already
> […](#)

Have already done that, thanks!

Listing available controls for device video0:
  Brightness
  Contrast
  Saturation
  White Balance Temperature, Auto
  Gain
  Power Line Frequency
  White Balance Temperature
  Sharpness
  Backlight Compensation
  Exposure, Auto
  Exposure (Absolute)
  Exposure, Auto Priority
  **Pan (relative)
  Tilt (relative)
  Pan Reset
  Tilt Reset**
  Focus
  LED1 Mode
  LED1 Frequency
  Disable video processing
  Raw bits per pixel

philsynek, 2022-03-19:
> It's not a matter of the controls being available, I just have to figure out a way to add them to the UI that makes sense 😄
> 
> The issue at hand is that these are relative controls and so a slider isn't really the right way to represent them. And the 'reset' ones will need to be buttons, but I'm not entirely sure how those actually work and without a camera to test with, that means I'll have to just guess and have some users test it out for me, provide feedback, fix and repeat.

Yep, that's why I am here! It will be a bit asynchronous, but I hope we will reach the goal.

philsynek, 2022-03-19:
Ok, so I testet a bit and what I achieved to do is:

pi@octopi:~ $ uvcdynctrl -s 'Pan Reset' 0
pi@octopi:~ $ uvcdynctrl -s 'Tilt Reset' 0
pi@octopi:~ $ uvcdynctrl -s 'Pan (relative)' -- -200
pi@octopi:~ $ uvcdynctrl -s 'Pan (relative)' 200

Those commands work!

The-EG, 2022-03-19:
> pi@octopi:~ $ uvcdynctrl -s 'Pan Reset' 0
> pi@octopi:~ $ uvcdynctrl -s 'Tilt Reset' 0
> pi@octopi:~ $ uvcdynctrl -s 'Pan (relative)' -- -200
> pi@octopi:~ $ uvcdynctrl -s 'Pan (relative)' 200

This is helpful, particularly the reset ones. 

I've hacked together something to test out.
@philsynek and anyone else that would like to help, please install this plugin from this URL:
https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/relative_controls.zip
(OctoPrint Settings->Plugin Manager->Get More->From URL)

With that version you should get multiple increase and decrease buttons for pan and tilt, and reset buttons for each. The increase and decrease buttons should range from 100%, 50% and 10% in both directions. Let me know how that works...maybe 25%, 10%, 1(%) would work better.

philsynek, 2022-03-19:
Great! Works basically. Only issue is: resets work only one time.

The-EG, 2022-03-20:
Ok, I think I've fixed the buttons not working more than once. Just reinstall from the same URL in my comment above again.

Also, does +100 move it all the way in that direction? Or can you press that more than once? I'm not sure what ranges would be most useful there.

perkinbr, 2022-03-20:
You folks were having so much fun, I thought I dust off the old octopi and try this out.

Looks good to me. +-100 works great on pan and is very near the limit.  For some reason mine seems to top top out at between +80-90 tilt, and bottom out at between -90-100. This one looks like it was abused a bit which I suppose could be a factor.

perkinbr, 2022-03-20:
Sorry, I realize that what I was saying wasn't totally clear. 
If you reset pan, you can move +100 or -100.
If you reset tilt, you can move +80 or -90 without hitting the stops.

philsynek, 2022-03-21:
Works like a charm! Reset does what it should every click.
My limits are:
-100 to +100 for pan
- 90 to +70 for tilt

The-EG, 2022-03-21:
Ok, thanks for the confirmation. I'll go through this again to clean it up a bit, then put it out as a release candidate for further testing.

perkinbr, 2022-03-25:
Tried out 0.4.0rc1 and it looks good.

commit ecd59328 (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 1b8b3d0c (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 20b8ce26 (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 7a4e4b39 (2022-03-21): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
closed 2022-05-13 by The-EG