The-EG/OctoPrint-CameraSettings #47: Wil not install for me.

alsmithson, 2021-06-28:
Installing plugin from https://github.com/The-EG/OctoPrint-CameraSettings/archive/main.zip...
/home/pi/oprint/bin/python2 -m pip --disable-pip-version-check install file:///tmp/tmpS8GkiX/main.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpS8GkiX/main.zip
OctoPrint-CameraSettings requires Python '>=3, <4' but the running Python is 2.7.13
Error!
Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

—-
Looks you require a newer version of Python than any other plugin I use. OctoPrint is working perfectly and I don't want to risk breaking it by installing a new version of Python.

The-EG, 2021-06-28:
You are running Python 2, which reached end of life status on January 1st, 2020.

This plugin only supports Python 3, which is what the error message indicates.

Please see the [FAQ](https://github.com/The-EG/OctoPrint-CameraSettings/blob/main/docs/faq.md#this-plugin-doesnt-show-up-in-the-plugin-manager-or-i-tried-to-install-it-manually-and-i-get-an-error-error-package-octoprint-camerasettings-requires-a-different-python-2716-not-in-3-4)  and the official guide on how to upgrade your OctoPrint installation to Python 3:
https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

> OctoPrint is working perfectly and I don't want to risk breaking it by installing a new version of Python.

While I understand the inconvenience, Python 3 is hardly new, OctoPrint officially supports it and since Python 2 is EOL, Py2 it is likely to receive less and less support as time goes on.

The-EG, 2021-06-28:
Duplicate of #45

closed 2021-06-28 by The-EG