The-EG/OctoPrint-CameraSettings #48: [FR] support for Logitech Webcam C270

robustini, 2021-06-28:
**Description of Desired Feature**
I thought that being a fairly common webcam it had already been integrated.
I have another RPI with the same configuration and a cheap webcam of which I don't know the brand, but in both I don't see the controls in the panel, and in the dropdown menu I can't select the webcams, no choice appears.

[image]

The-EG, 2021-06-28:
This isn't an issue with compatibility; many users have Logitech cameras and I'm fairly certain I've seen the C270 mentioned at some point. Since you aren't running OctoPi, my first thought is that this is probably a permissions issue.

The user running OctoPrint (definitely not root unless you purposely disregarded all of the warnings) will need access to the `/dev/video*` devices.

First, check what group owns them (`ls -l /dev/video*`):

``` terminal
taylor@printerpi:~ $ ls -l /dev/video*
crw-rw----+ 1 root video 81, 0 Jun 25 22:10 /dev/video0
crw-rw----+ 1 root video 81, 1 Jun 25 22:10 /dev/video1
crw-rw----+ 1 root video 81, 2 Jun 25 22:02 /dev/video10
crw-rw----+ 1 root video 81, 3 Jun 25 22:02 /dev/video11
crw-rw----+ 1 root video 81, 4 Jun 25 22:02 /dev/video12
crw-rw----+ 1 root video 81, 5 Jun 25 22:02 /dev/video13
crw-rw----+ 1 root video 81, 6 Jun 25 22:02 /dev/video14
crw-rw----+ 1 root video 81, 7 Jun 25 22:02 /dev/video15
crw-rw----+ 1 root video 81, 8 Jun 25 22:02 /dev/video16
crw-rw----+ 1 root video 81, 9 Jun 25 22:02 /dev/video2
taylor@printerpi:~ $
```

It's probably `video`, but it doesn't hurt to verify. Also, you want to make sure that the group is able to access the devices. This is the second `rw` in the mode line, `crw-rw----+`. If that says `crw-------+` or even `crw-r-----+` then even being in the correct group wouldn't help. If these are wrong it will involve some more research to fix; you are probably using udev and it will require changing some configuration files.

Then, add the appropriate user to the group: `sudo usermod -a -G video username`, where `video` is the group mentioned above from `ls` and `username` is the user that runs OctoPrint.

The-EG, 2021-06-28:
Just to be sure, I went back looking through some earlier FR's, and I did indeed have a user using a C270 already: #23

So the camera is definitely supported and should already have all controls implemented on the form.

robustini, 2021-06-28:
> 
> 
> This isn't an issue with compatibility; many users have Logitech cameras and I'm fairly certain I've seen the C270 mentioned at some point. Since you aren't running OctoPi, my first thought is that this is probably a permissions issue.
> 
> The user running OctoPrint (definitely not root unless you purposely disregarded all of the warnings) will need access to the `/dev/video*` devices.
> 
> First, check what group owns them (`ls -l /dev/video*`):
> 
> ```
> taylor@printerpi:~ $ ls -l /dev/video*
> crw-rw----+ 1 root video 81, 0 Jun 25 22:10 /dev/video0
> crw-rw----+ 1 root video 81, 1 Jun 25 22:10 /dev/video1
> crw-rw----+ 1 root video 81, 2 Jun 25 22:02 /dev/video10
> crw-rw----+ 1 root video 81, 3 Jun 25 22:02 /dev/video11
> crw-rw----+ 1 root video 81, 4 Jun 25 22:02 /dev/video12
> crw-rw----+ 1 root video 81, 5 Jun 25 22:02 /dev/video13
> crw-rw----+ 1 root video 81, 6 Jun 25 22:02 /dev/video14
> crw-rw----+ 1 root video 81, 7 Jun 25 22:02 /dev/video15
> crw-rw----+ 1 root video 81, 8 Jun 25 22:02 /dev/video16
> crw-rw----+ 1 root video 81, 9 Jun 25 22:02 /dev/video2
> taylor@printerpi:~ $
> ```
> 
> It's probably `video`, but it doesn't hurt to verify. Also, you want to make sure that the group is able to access the devices. This is the second `rw` in the mode line, `crw-rw----+`. If that says `crw-------+` or even `crw-r-----+` then even being in the correct group wouldn't help. If these are wrong it will involve some more research to fix; you are probably using udev and it will require changing some configuration files.
> 
> Then, add the appropriate user to the group: `sudo usermod -a -G video username`, where `video` is the group mentioned above from `ls` and `username` is the user that runs OctoPrint.

[image]

The-EG, 2021-06-28:
Ok, go to Logging in the OctoPrint settings window and enable debug logging for this plugin:
[image]

Save settings (close the settings window) then try to use the plugin again. Then go back to the logging tab, download `octoprint.log` and attach it here.

You can then delete the debug entry if you want.

robustini, 2021-06-29:
This is the log of this cam: https://www.amazon.co.uk/dp/B0899RJQ95/?coliid=ILORCON7SEZDB&colid=ZMP5PR7IDTC5&psc=1&ref_=lv_ov_lig_dp_it

[octoprint-logs.zip](https://github.com/The-EG/OctoPrint-CameraSettings/files/6731075/octoprint-logs.zip)

robustini, 2021-06-29:
And this one of the other rpi with the C270, same issue.

[octoprint-logs.zip](https://github.com/The-EG/OctoPrint-CameraSettings/files/6731105/octoprint-logs.zip)

robustini, 2021-06-29:
If I go directly to mjpg streamer as you can see the cam controls work.

[image]

[image]

robustini, 2021-06-29:
I'm noticing this though, the control url has no "webcam" in the address, could this be the problem?

[image]

robustini, 2021-06-29:
pi@Raspberry2:~ $ ps aux | grep stream                                                                      root       366  2.1  1.1  72980 10896 ?        Sl   07:55   1:02 /usr/local/bin/mjpg_streamer -i /usr/local/lib/mjpg-streamer/input_uvc.so -d /dev/video0 --fps 30 -q 85 -r 1280x720  -o /usr/local/lib/mjpg-streamer/output_http.so -p 8085 -w /usr/local/share/mjpg-streamer/www -b
pi        5790  0.0  0.0   4368   528 pts/1    S+   08:43   0:00 grep --color=auto stream
pi@Raspberry2:~ $ v4l2-ctl --list-devices
bcm2835-codec (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

JOYACCESS JA-Webcam: JOYACCESS  (usb-3f980000.usb-1.2):
        /dev/video0
        /dev/video1

pi@Raspberry2:~ $

robustini, 2021-06-29:
And with Logitech C270:

pi@Raspberry1:~ $ ps aux | grep stream

root       504  1.6  0.9  82280  8924 ?        Sl   07:55   0:50 /usr/local/bin/mjpg_streamer -i /usr/local/lib/mjpg-streamer/input_uvc.so -d /dev/video0 --fps 15 -q 85 -r 1280x960  -o /usr/local/lib/mjpg-streamer/output_http.so -p 8080 -w /usr/local/share/mjpg-streamer/www -b
pi        8296  0.0  0.0   4368   568 pts/1    S+   08:45   0:00 grep --color=auto stream

pi@Raspberry1:~ $ v4l2-ctl --list-devices

bcm2835-codec (platform:bcm2835-codec):
        /dev/video10
        /dev/video11
        /dev/video12

UVC Camera (046d:0825) (usb-3f980000.usb-1.4):
        /dev/video0
        /dev/video1

robustini, 2021-06-29:
If your plugin uses v4l2-ctl to intercept the cams and be able to configure them I don't understand why it doesn't see them, since the v4l2-ctl command responds with all the settings and specs.

The-EG, 2021-06-29:
Yes, it's actually running `v4l2-ctl` in the background for many things.

From the log, it's not getting any errors but it's also not finding any useful cameras for some reason:

```text
2021-06-29 08:04:44,179 - octoprint.plugins.camerasettings - DEBUG - Building camera list
2021-06-29 08:04:44,182 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video12
2021-06-29 08:04:44,205 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video1
2021-06-29 08:04:44,229 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video10
2021-06-29 08:04:44,252 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video11
2021-06-29 08:04:44,276 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video0
2021-06-29 08:04:44,301 - octoprint.plugins.camerasettings - DEBUG - Cameras found: []
```

I'll have to go dig a bit deeper and see how to debug this further.
 
In the meantime, does the output of `v4l2-ctl --list-ctrls-menus` produce the expected output (should be lists of the cameras controls, similar to what you have above)?

Also, I see you are running mjpg-streamer as root...

robustini, 2021-06-29:
This is the output of the JOYACCESS JA-Webcam:

pi@Raspberry2:~ $ v4l2-ctl --list-ctrls-menus

                     brightness (int)    : min=-64 max=64 step=1 default=0 value=0
                       contrast (int)    : min=0 max=64 step=1 default=32 value=32
                     saturation (int)    : min=0 max=128 step=1 default=64 value=64
                            hue (int)    : min=-40 max=40 step=1 default=0 value=0
 white_balance_temperature_auto (bool)   : default=1 value=1
                          gamma (int)    : min=72 max=500 step=1 default=100 value=100
                           gain (int)    : min=0 max=100 step=1 default=0 value=0
           power_line_frequency (menu)   : min=0 max=2 default=1 value=1
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature (int)    : min=2800 max=6500 step=1 default=4600 value=4600 flags=inactive
                      sharpness (int)    : min=0 max=6 step=1 default=3 value=3
         backlight_compensation (int)    : min=0 max=2 step=1 default=1 value=1
                  exposure_auto (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
              exposure_absolute (int)    : min=1 max=5000 step=1 default=157 value=157 flags=inactive
         exposure_auto_priority (bool)   : default=0 value=1

If mjpg-streamer it starts as root I didn't force it, it's like this in Repetier Server: https://www.repetier-server.com/setting-webcam-repetier-server-linux/

The-EG, 2021-06-29:
> If mjpg-streamer it starts as root I didn't force it.

Understood, but it also means that mjpg-streamer working doesn't really indicate a lack of permissions issues because running it as root will completely bypass everything.

The-EG, 2021-06-30:
I haven't had a chance to figure out where I can put more debug logging in to look at this further...but I was thinking about this today and:
- Does `v4l2-ctl --set-ctrl brightness=10` work without sudo? Can you change brightness, contrast, etc from the shell as `pi`?
- Have you restarted OctoPrint since adding pi to the video group?
- Next time you can, can you try restarting the pi? It *shouldn't* be necessary, although restarting OctoPrint would be...
 
Everything I'm seeing points to the `v4l2-ctl --list-ctrls-menus` command as running but not producing the expected output. There's only 2 things that I can think of that would cause that:
- permissions issues (you can't write to the video devices)
- something weird with the output of `v4l2-ctl` itself or the way Python is parsing that output (is your locale set to something exotic?)

robustini, 2021-07-01:
>     * Does `v4l2-ctl --set-ctrl brightness=10` work without sudo? Can you change brightness, contrast, etc from the shell as `pi`?

Yes, tried now, work on both my cams on both my rpi without sudo.
 
>     * Have you restarted OctoPrint since adding pi to the video group?

Sure!

>     * Next time you can, can you try restarting the pi? It _shouldn't_ be necessary, although restarting OctoPrint would be...

Done but does not give any results regarding the problem.
The controls work, it is your plugin that not finding the cam obviously does not activate the sliders for the controls. 
Not understanding how your plugin intercepts the cam I can't help you that much.
If "v4l2-ctl --all" correctly shows the device, with its ID and much more, why does your plugin not rely on that output to assign the name of the device found in the drop down menu?

Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : **UVC Camera (046d:0825)**
        Bus info      : usb-3f980000.usb-1.4
        Driver version: 4.19.66
        Capabilities  : 0x84A00001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)

The-EG, 2021-07-01:
> Wouldn't it be enough for your plugin to intercept the "card type" ( UVC Camera (046d:0825) ) and assign that name in the drop down menu?

If only it were that simple 😄.

First of all, that name is not always something that would make sense to a typical user. In the case of a raspberry pi camera:

```terminal
Driver Info:
        Driver name      : bm2835 mmal
        Card type        : mmal service 16.1
        Bus info         : platform:bcm2835-v4l2-0
        Driver version   : 5.10.17
        Capabilities     : 0x85200005
```

Further, as you may have noticed by now, you have quite a few more video devices than actual cameras. Some of them have to do with the video processing capabilities of the RPi's GPU and other times a camera will actually create multiple devices, but only one is valid for video capture.

So, the only way to figure out what is a camera and what isn't is to enumerate them and look at their properties. The most straight forward thing I found was to look for devices that can be controlled. In pretty much every case, the camera ends up being the only thing with controls.

You may be thinking, "I only get one device when I run `v4l2-ctl`." Of course you do, it's defaulting to /dev/video0. You can specify the device (ie `-d /dev/video1`) and certain arguments will use all devices (`--list-devices`).

Why can't we just use `/dev/video0` by default and call it a day? That may not always be the correct device, especially in the case where a camera creates multiple video devices. Or a user could have multiple cameras hooked up to a single raspberry pi. This is actually pretty common, even my development environment has this case. 

So, no, the plugin must find *all* video devices that are actually cameras and let the user decide which one to control. 

But, all this means that I have to go poking around quite a bit into *just* to get a list of cameras. Not so simple.

And, even if I were to just disregard all of that and always use `/dev/video0` and pull in the card name instead....it *still* wouldn't populate the controls because *that* command appears to be where it's failing.

robustini, 2021-07-01:
Repetier Server, which is in the same RPI (obviously stopped if I use OctoPrint) offers only two devices, in fact they are the only two that are created by the cam.
So in that case a distinction from the other "videos" is made. 

[image]

The-EG, 2021-07-01:
Those are the devices in `/dev/v4l2/by-id`. That will not work for RaspiCams/AruCams etc.

robustini, 2021-07-01:
I understand, so give me a chance to tell the plugin which one to use (in my case).

The-EG, 2021-07-01:
I appreciate you trying to offer what you consider to be a simple solution, but as long as the plugin can't read the output of `v4l2-ctl --list-ctrls-menus`, you won't be able to use it.

And, that would additional complexity to this that isn't needed because if the above is fixed it will work as intended.

The-EG, 2021-07-01:
> Not everyone uses OctoPi.

This is exactly why I have structured things the way they are. Not all linux distributions create the devices symlinked in `/dev/v4l/by-id` or even have udev! Also, I not assuming mjpg-streamer is in use either! In fact, you are free to use whatever streamer you wish, I use uStreamer personally.

robustini, 2021-07-01:
Thanks anyway for making me understand why it doesn't work in my case.
I remove your plugin and go the old way using mjpg-streamer settings, once found they don't touch them anymore.

The-EG, 2021-07-01:
Ok.

For reference, in case this comes up again, etc.: I just finished adding some additional debugging. It can be installed from:
https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/dev/debug_cameralist.zip

robustini, 2021-07-01:
[octoprint-logs.zip](https://github.com/The-EG/OctoPrint-CameraSettings/files/6747834/octoprint-logs.zip)

The-EG, 2021-07-01:
Well, here's the problem:

```text
2021-07-01 13:47:15,316 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video0
2021-07-01 13:47:15,366 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                      brightness (int)    : min=0 max=255 step=1 default=128 value=128
2021-07-01 13:47:15,366 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                        contrast (int)    : min=0 max=255 step=1 default=32 value=32
2021-07-01 13:47:15,366 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                      saturation (int)    : min=0 max=255 step=1 default=32 value=32
2021-07-01 13:47:15,367 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:  white_balance_temperature_auto (bool)   : default=1 value=1
2021-07-01 13:47:15,367 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                            gain (int)    : min=0 max=255 step=1 default=64 value=131
2021-07-01 13:47:15,367 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:            power_line_frequency (menu)   : min=0 max=2 default=2 value=2
2021-07-01 13:47:15,368 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 				0: Disabled
2021-07-01 13:47:15,368 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 				1: 50 Hz
2021-07-01 13:47:15,368 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 				2: 60 Hz
2021-07-01 13:47:15,369 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:       white_balance_temperature (int)    : min=0 max=10000 step=10 default=4000 value=6610 flags=inactive
2021-07-01 13:47:15,369 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                       sharpness (int)    : min=0 max=255 step=1 default=24 value=24
2021-07-01 13:47:15,370 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:          backlight_compensation (int)    : min=0 max=1 step=1 default=0 value=0
2021-07-01 13:47:15,371 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:                   exposure_auto (menu)   : min=0 max=3 default=3 value=3
2021-07-01 13:47:15,372 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 				1: Manual Mode
2021-07-01 13:47:15,372 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 				3: Aperture Priority Mode
2021-07-01 13:47:15,372 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:               exposure_absolute (int)    : min=1 max=10000 step=1 default=166 value=664 flags=inactive
2021-07-01 13:47:15,373 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match:          exposure_auto_priority (bool)   : default=0 value=1
2021-07-01 13:47:15,373 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no CTRL_PAT match: 
```

For reference, this is what one of my cameras produces:
```text
2021-07-01 07:32:22,696 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video0
2021-07-01 07:32:22,760 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                      brightness 0x00980900 (int)    : min=1 max=255 step=1 default=128 value=128
2021-07-01 07:32:22,760 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                        contrast 0x00980901 (int)    : min=1 max=255 step=1 default=128 value=128
2021-07-01 07:32:22,761 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                      saturation 0x00980902 (int)    : min=1 max=255 step=1 default=128 value=90
2021-07-01 07:32:22,762 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:  white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=0
2021-07-01 07:32:22,769 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                            gain 0x00980913 (int)    : min=1 max=100 step=1 default=50 value=50
2021-07-01 07:32:22,771 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:            power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=1 value=1
2021-07-01 07:32:22,773 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               0: Disabled
2021-07-01 07:32:22,774 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               1: 50 Hz
2021-07-01 07:32:22,784 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               2: 60 Hz
2021-07-01 07:32:22,785 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:       white_balance_temperature 0x0098091a (int)    : min=2800 max=6500 step=1 default=4650 value=6500
2021-07-01 07:32:22,786 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                       sharpness 0x0098091b (int)    : min=1 max=255 step=1 default=128 value=128
2021-07-01 07:32:22,787 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                   exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=0 value=0
[... cut ...]
2021-07-01 07:32:22,803 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:          exposure_auto_priority 0x009a0903 (bool)   : default=0 value=0
2021-07-01 07:32:22,807 - octoprint.plugins.camerasettings - DEBUG - [video0] Skipping line, no MENU_PAT match:
```

The version of `v4l2-utils` isn't ouputting the control id. That's fine, this plugin doesn't need it but this is the first time I've seen it formatted in that way in nearly 3,000 installs. That should be fairly easy to fix

robustini, 2021-07-01:
> on your system isn't ouputing the control id

If your plugin normally ignores those IDs, sure it's easy to fix it!

The-EG, 2021-07-01:
> If your plugin normally ignores those IDs, sure it's easy to fix it!

Indeed, just had to adjust a regular expression.

If you'd like to try it, install: https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fixes/v4l2-utils-formatting.zip

robustini, 2021-07-01:
I believe you can put it on the master, I'm glad I helped you make your plugin better, thanks! 

[image]

closed 2021-07-01 by robustini

The-EG, 2021-07-01:
Thanks, glad to see it working.

> I'm glad I helped you make your plugin better

Indeed you have, that's why I was so adamant about fixing it rather than a work around, haha. I'll also add the 'bcm2835-codec' devices to the ignored list too.

robustini, 2021-07-01:
0.21 result, damn! :-)

[image]

The-EG, 2021-07-01:
> No no, false alarm, everything ok!

I was just typing this 😄: 
Close the settings window and try it again. If you happen to open it up when OctoPrint has just started or even when the web browser is still connecting to the server the messages get 'lost' sometimes...I've actually debugged it quite a bit to see if there's anything I can do, but it's something going on between OctoPrint sending the message and the browser receiving it. Long story short, don't open the settings window until it's fully connected or just try again!

VanillaChief, 2021-07-07:
I am experiencing a similar problem. I am using a Logitech CS920 that is directly attached to the OctoPi device. 
I can manage the output of the Webcam directly with v4l2-ctl via SSH but do not get any controls in the plugin. 
I've read the whole thread and tried to implement the different suggested solutions and also update camera-settings to 0.2.1 but had no success. 

The Logs show an Error but I am not sure if that is the problem:

`2021-07-07 16:09:42,061 - octoprint.plugins.logging - INFO - Setting logger octoprint.plugins.camerasettings level to 10
2021-07-07 16:10:53,059 - octoprint.plugins.camerasettings - DEBUG - Building camera list
2021-07-07 16:10:53,061 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture0 based on ^bcm2835-isp-\w+$
2021-07-07 16:10:53,062 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-isp based on ^bcm2835-codec-\w+$
2021-07-07 16:10:53,063 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video1
2021-07-07 16:10:53,089 - octoprint.plugins.camerasettings - DEBUG - [video1] Skipping line, no CTRL_PAT match:
2021-07-07 16:10:53,091 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-decode based on ^bcm2835-codec-\w+$
2021-07-07 16:10:53,093 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture1 based on ^bcm2835-isp-\w+$
2021-07-07 16:10:53,098 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-output0 based on ^bcm2835-isp-\w+$
2021-07-07 16:10:53,099 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-encode based on ^bcm2835-codec-\w+$
2021-07-07 16:10:53,101 - octoprint.plugins.camerasettings - DEBUG - Getting controls for video0
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                      brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                        contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                      saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:  white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                            gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=109
2021-07-07 16:10:53,130 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:            power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               0: Disabled
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               1: 50 Hz
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               2: 60 Hz
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:       white_balance_temperature 0x0098091a (int)    : min=2000 max=6500 step=1 default=4000 value=4763 fl$
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                       sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:          backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
2021-07-07 16:10:53,131 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                   exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               1: Manual Mode
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found MENU item in:                               3: Aperture Priority Mode
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:               exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=83 flags=in$
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:          exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                    pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                   tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                  focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=50 flags=inact$
2021-07-07 16:10:53,132 - octoprint.plugins.camerasettings - DEBUG - [video0] Found CONTROL in:                      focus_auto 0x009a090c (bool)   : default=1 value=1
[... cut ...]
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_CameraSettings/__init__.py", line 102, in on_api_command
    self.do_cameras_list_event()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_CameraSettings/__init__.py", line 180, in do_cameras_list_event
    if (cam_map[c]['camera'] not in cam_names): del cam_map[c]
IndexError: list index out of range`

robustini, 2021-07-07:
> I am experiencing a similar problem. I am using a Logitech CS920 that is directly attached to the OctoPi device.
> I can manage the output of the Webcam directly with v4l2-ctl via SSH but do not get [...]

From what you can see from your log, very similar to mine, 0.21 should work with your cam instead.

robustini, 2021-07-07:
> I am experiencing a similar problem. I am using a Logitech CS920 that is directly attached to the OctoPi device.
> I can manage the output of the Webcam directly with v4l2-ctl via SSH but do not get [...]

From what i can see from your log, very similar to mine, 0.21 should work with your cam instead.
Maybe that cam has some control that is not covered by the plugin, from what I see, however, all the controls seem to be covered, so very strange.

VanillaChief, 2021-07-07:
I could try to reset my octopi.txt. I still have it configured to my rpi-cam I had before the CS920. I cannot think of anything else to change. 
Will report back when I've tried it, currently the printer is printing.

The-EG, 2021-07-07:
It's a different error. I'm taking a look at it

The-EG, 2021-07-07:
> I could try to reset my octopi.txt. I still have it configured to my rpi-cam I had before the CS920. I cannot think of anything else to change.
> Will report back when I've tried it, currently the printer is printing.

@VanillaChief No need to do that, it shouldn't have any effect. I've opened a new issue for this since it's different and I'll see what's going on.

commit cff62547 (2021-07-01): Update control parsing regexp

Do not expect v4l2-ctl to output control id.
Fix #48