The-EG/OctoPrint-CameraSettings #49: [BUG] Python error in get_cameras

The-EG, 2021-07-07:
From #48:

> I am experiencing a similar problem. I am using a Logitech CS920 that is directly attached to the OctoPi device.
> I can manage the output of the Webcam directly with v4l2-ctl via SSH but do not get any controls in the plugin.
> I've read the whole thread and tried to implement the different suggested solutions and also update camera-settings to 0.2.1 but had no success.
>
> The Logs show an Error, but I am not sure if that is the problem: https://gist.github.com/VanillaChief/6e10e45b63d6a98f8deb883cd595fe93

The-EG, 2021-07-07:
@vanillaChief Do you have the 'MultiCam' plugin installed? If so, what version?

PS. this is why full logs are helpful and why plugin/octoprint devs ask for them. I understand the aversion to providing something you think may have personal information in it, but octoprint.log *shouldn't* have anything overly sensitive and it would have provided the answer to the above two questions plus a few more that I'm probably going to ask next.

VanillaChief, 2021-07-07:
here is the complete log: https://gist.github.com/VanillaChief/e7c68ba1e3a69e4a169da3ecede5fd22

I don't have the MultiCam Plugin installed.

The-EG, 2021-07-07:
Thank you. Did you have multiple different cameras plugged in at some point? I think I see what's going on....

In the meantime, can you try this: over ssh, run
1. `sudo systemctl stop octoprint`
2. `~/oprint/bin/octoprint config remove plugins.camerasettings.multicam_mapping`
3. `sudo systemctl start octoprint`

Note, this will stop and then restart OctoPrint so only try when it's safe to do that.

VanillaChief, 2021-07-07:
yeah, I tried a couple webcams. Will try your solution once my print has finished.

The-EG, 2021-07-07:
Actually, I have something easier to try: install this version - https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fixes/issue49.zip

You found a bug in how the multicam mapping is filtered out and in the process I found a bug in some of the debug logging...two for one 😄.

Even when you don't have multicam enabled, it builds an 'empty' mapping (even for one camera) because it's much simpler than trying to figure out if the plugin is installed and if multiple cameras are present, etc.

I think, in your case you had a prior camera that now didn't exist (it was unplugged) and it was trying to update that mapping and failed. It should work properly now in any case.

VanillaChief, 2021-07-08:
That did the trick. Installed the fix, and it works now on first try. Thanks!

The-EG, 2021-07-08:
Thanks, released in 0.2.2.

commit 8bca212f (2021-07-08): Build multicam_mapping without removing nonexistant items

Fix #49
closed 2021-07-08 by The-EG