The-EG/OctoPrint-CameraSettings #50: [FR] Implement Controls for Logitech QuickCam Orbit MP aka: "UCV Camera (046d:08ad)"

ethzero, 2021-08-11:
Reference: 
https://support.logi.com/hc/en-gb/articles/360025260794--Product-Gallery-QuickCam-Orbit-MP
https://support.logi.com/hc/en-gb/articles/360023307254-QuickCam-Orbit-MP-Technical-Specifications

PS: This is a insanely old camera (circa 2007). From personal experience, it's next to impossible using very old Windows drivers to even get an image out of this camera, and yet, Linux has **no problem**. PTZ would be a nice-to-have but I understand from old Linux driver posts that it's tricky.

{
  "pan_relative": {
    "type": "int",
    "min": "-4480",
    "max": "4480",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  },
  "tilt_relative": {
    "type": "int",
    "min": "-1920",
    "max": "1920",
    "step": "0",
    "default": "0",
    "value": "0",
    "flags": "write"
  },
  "pan_reset": {
    "type": "button",
    "flags": "write"
  },
  "tilt_reset": {
    "type": "button",
    "flags": "write"
  }
}

The-EG, 2021-08-11:
Hi there, you aren't the only one using one of these cameras, actually... I was discussing the pan and tilt controls in #46 . 

I believe that pan and tilt will function, if you want to try the branch mentioned there you are welcome to. The issue is that they should apparently be 'buttons,' where the given value is added or subtracted to the current value, but V4L represents them as absolute values.

And, of course, there are actual buttons for the 'reset.'

I'm willing to try to develop these 'button' controls but it's going to require some trial and error. Most of the time this wouldn't be an issue, but these are fairly rare and none of the cameras I have access to have them. So I'll either need to try to get something similar, or I'll have to work with a user that's willing to do some testing for me.

The-EG, 2021-08-11:
duplicate of #46

commit ecd59328 (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 1b8b3d0c (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 20b8ce26 (2022-03-20): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
commit 7a4e4b39 (2022-03-21): ✨ Support Relative and Button Controls

Fixes #42
Fixes #46
Fixes #50
closed 2022-03-20 by The-EG