The-EG/OctoPrint-CameraSettings #53: [BUG] No preview when using HLS stream type

pandel, 2021-09-10:
Hi!

I am using your plugin with the newest OctoPi release and set "camera_streamer=hls" (octopi.txt). This gives a much better video performance in my network and lower overall system load.

Sadly, the stream is not shown in the preview area of your plugin. Would be nice, if you'd take look at this...

Thanks, stay safe!

Regards,
Holger

The-EG, 2021-09-10:
I'll have to review this to see how OctoPrint deals with the HLS stream. The plugin doesn't know or even care what program is streaming or what type of stream it is. The preview is just showing the same URL that's configured in OctoPrint but I suspect there must be something a bit different going on there to work with HLS.

pandel, 2021-09-10:
Thanks! I was wondering also about why the stream is not showing up. I looked through your code, but did not find anything special that could perhaps prevent it...

I made a test: I copied the HTML tag from UI customizers video frame via the browser console to your plugin view and replaced the existing img tag with:

`<video muted="" autoplay="" style="width: 100%" src="/hls/stream.m3u8"></video>`

That did the trick ;-) - video is shown.

The-EG, 2021-09-10:
Good to know you can just put it in a video tag. The problem is the plugin has to have a way of knowing the stream should be hls. I may just put in a generic option to use a video element instead of a img.

The-EG, 2021-09-11:
> That did the trick ;-) - video is shown.

Which browser are you using? Just using a video tag doesn't work in my environment:
[image]
[image]

Do you also have a plugin installed to change the main OctoPrint camera view to HLS? It may be doing more than just that...

pandel, 2021-09-11:
I am using Safari on Mac OS.

No, I don't used any plugin. I changed the `camera_streamer`option in octopi.txt and rebooted the Pi. After that, I only changed some values in `/etc/systemd/system/ffmpeg_hls.conf` regarding resultion, etc. . In Octoprint, I set the stream source to `hls/stream.m3u8`.

Via browser console, I changed the your img element to this:
`<video muted="" autoplay="" style="width: 100%" src="/hls/stream.m3u8"></video>`

The-EG, 2021-09-11:
Ok, it's sounding like support various across different browsers. This is going to take more effort than just a video tag, unfortunately.

The-EG, 2021-09-24:
Ok, I poked around at how OctoPrint is supporting HLS streams and I've replicated the same thing...this should work for any browser that the main 'Control' view with HLS streams currently works with.

Please try it out, install this plugin with this URL https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fr/hls-preview.zip

I'll get this into an RC soon and hopefully get some more people to test it too.

The-EG, 2021-09-25:
Fix now available in 0.3.0rc1

commit 883a4a70 (2021-09-25): Support HLS stream

fixes #53
closed 2021-09-29 by pandel