The-EG/OctoPrint-CameraSettings #54: [BUG] New Issue

S7EN, 2021-09-21:
**Observed behavior**
Focus Absolute Settings get not saved

**Expected behavior**
Should be saved

**OctoPrint Version**
Latest

**CameraSettings Plugin Version**
Latest

**Camera**
Logitech C920

**Linux/OctoPi**
OctoPi latest

**Other Plugins**
No

The-EG, 2021-09-21:
When you say 'not saved' do you mean it's not taking affect or it's not saving in the preset?

S7EN, 2021-09-21:
When i saved the settings and restart the pi the focus absolute is 0 again. Tried to delete the preset and reinstalled the plugin but it takes no effect.

The-EG, 2021-09-21:
Increment `'Startup Preset Apply Count` and try it again.

See https://github.com/The-EG/OctoPrint-CameraSettings/blob/main/docs/faq.md#not-all-settings-are-applied-on-on-startup

S7EN, 2021-09-21:
Ah okay, didn`t know about the faq and solution. That worked, thanks.

closed 2021-09-21 by S7EN

The-EG, 2021-09-21:
> Ah okay, didn`t know about the faq and solution. That worked, thanks.

No problem, I'm glad that's all it was, and thanks for confirming!