The-EG/OctoPrint-CameraSettings #55: [FR] Implement Controls for Microsoft LifeCam HD-3000

BlueWings172, 2021-09-28:
Hello

Just installed this plugin and I can't find any settings to control. I have Octoprint on Docker (IOTstack).

Is this plugin compatible with Docker ? Is the issue that my webcam model (Microsoft LifeCam HD-3000) not supported?

[image]

This is what is see on the plugin page on Octoprint. Stream loads fine but the dropdown menu next it does not show any items when I click on it and there are no settings to control.

I would appreciate any help.

Thanks

The-EG, 2021-09-28:
Hi there, I'm not sure about the compatibly with docker, but in general the environment that OctoPrint runs in would need to have access to the camera. Specially, if the applicable `/dev/videoN` device isn't available

The-EG, 2021-09-28:
Also, your camera should work. In addition to the device I mentioned above, the plugin looks at some specific sysfs files to look for the camera names, so they'd have to be available too.

closed 2021-09-28 by The-EG

BlueWings172, 2021-09-28:
> Hi there, I'm not sure about the compatibly with docker, but in general the environment that OctoPrint runs in would need to have access to the camera. Specially, if the applicable `/dev/videoN` device isn't available then the plugin won't work.

Octoprint does already have access to my camera and the camera works well with plugins like Octolapse and Sidebar Webcam. 
In the screenshot I have included above, you can see the webcam streaming.

The-EG, 2021-09-28:
No, OctoPrint only needs access to the camera *URL* for those things to work. This plugin works independently of the steamer program, and so the the OctoPrint process needs access to the actual camera device at the operating system level.

Running OctoPrint in a docker container had likely isolated it away from that access and resolving that is outside of the scope of this plugin or what I can help with.