The-EG/OctoPrint-CameraSettings #60: Implement Controls for MP0023 KEYESTUDIO 5 Megapixels Camera Module Customized

babsevensix, 2021-11-13:
SMP0023 KEYESTUDIO 5 Megapixels Camera Module Customized

{
  "camera_orientation": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Front"
      },
      {
        "value": "1",
        "desc": "Back"
      },
      {
        "value": "2",
        "desc": "External"
      }
    ],
    "min": "0",
    "max": "2",
    "default": "2",
    "value": "2",
    "flags": "read"
  },
  "camera_sensor_rotation": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "1",
    "default": "0",
    "value": "0",
    "flags": "read"
  },
  "vertical_blanking": {
    "type": "int",
    "min": "24",
    "max": "32287",
    "step": "1",
    "default": "24",
    "value": "509"
  },
  "horizontal_blanking": {
    "type": "int",
    "min": "1212",
    "max": "1212",
    "step": "1",
    "default": "1212",
    "value": "1212",
    "flags": "read"
  },
  "analogue_gain": {
    "type": "int",
    "min": "16",
    "max": "1023",
    "step": "1",
    "default": "32",
    "value": "96"
  },
  "pixel_rate": {
    "type": "int64",
    "min": "55000000",
    "max": "55000000",
    "step": "1",
    "default": "55000000",
    "value": "55000000",
    "flags": "read"
  }
}

The-EG, 2021-11-13:
Oh, exciting! It's been a while since a new camera has come through, I was wondering if I somehow had them all...

Many of these look like they are read only or otherwise not something you can adjust. but a couple look to be usable. I'll add everything in the background to get the warning to go away, but I'm only going to add the ones that look like they can actually be set to the UI.

Shouldn't take long, I'll try to get it within a day or so.

The-EG, 2021-11-21:
Ok, so I was a bit optimistic when I thought I'd have a chance to look at this last week....but, it is a quick fix at least and I was able to do it this morning.

Please take a look at it, you can try it out by installing the plugin using this URL:
https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/heads/fixes/fr-60.zip

As I mentioned, I'm not sure that everything will be editable, but this will at least expose them if they are useful and get rid of the warning message.

babsevensix, 2021-11-23:
Thanks for now. I have tried the changes, in fact there is no more message At
the moment, however, I see this:
[image: immagine.png]

commit 492c08c8 (2021-11-23): ✨ Add more camera controls (fix #60)
closed 2022-01-11 by The-EG