The-EG/OctoPrint-CameraSettings #62: [FR] Implement Controls for Raspberri Pi camera v2

erripe13, 2021-11-27:
**Camera manufacturer and model (ie. NexiGO N660)**:
Raspberry pi camera v2

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "force_key_frame": {
    "type": "button",
    "flags": "write"
  },
  "h264_minimum_qp_value": {
    "type": "int",
    "min": "0",
    "max": "51",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "h264_maximum_qp_value": {
    "type": "int",
    "min": "0",
    "max": "51",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

The-EG, 2021-11-28:
These controls are now available for testing in version 0.3.1rc2.

You can update to that by selecting the Release Candidate channel for this plugin:
[image]

erripe13, 2021-11-29:
Working ! 😄

commit 7ab178f4 (2021-11-28): ✨ Add more camera controls (fix #62)
closed 2022-01-11 by The-EG