The-EG/OctoPrint-CameraSettings #65: [FR] Implement Controls for [Camera Name]

Thegs68, 2022-01-08:
**Camera manufacturer and model (ie. NexiGO N660)**:
Day-Night V2 for Raspberry Pi Camera, Arducam All-Day Image All-Model Support, IR LED for Low Light and Night Vision, M12 Lens Interchangeable, IR Filter Auto-Switch Programmable, OV5647 5MP 1080P

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "force_key_frame": {
    "type": "button",
    "flags": "write"
  },
  "h264_minimum_qp_value": {
    "type": "int",
    "min": "0",
    "max": "51",
    "step": "1",
    "default": "0",
    "value": "0"
  },
  "h264_maximum_qp_value": {
    "type": "int",
    "min": "0",
    "max": "51",
    "step": "1",
    "default": "0",
    "value": "0"
  }
}

The-EG, 2022-01-08:
This is actually already addressed in the latest pre-release. I guess this might be a hint to finally release that...

duplicate of #62

closed 2022-01-08 by The-EG

Thegs68, 2022-01-08:
Thanks for your prompt reply, I will try the RC.