The-EG/OctoPrint-CameraSettings #69: [FR] Implement Controls for [Camera Name]

danvandermeir, 2022-02-10:
**Camera manufacturer and model (ie. NexiGO N660)(imx559 Arducam Autofocus High Resolution)**:

**Unknown controls information **:
{
  "red_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "green_red_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "blue_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "green_blue_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "test_pattern": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Disabled"
      },
      {
        "value": "1",
        "desc": "Color Bars"
      },
      {
        "value": "2",
        "desc": "Solid Color"
      },
      {
        "value": "3",
        "desc": "Grey Color Bars"
      },
      {
        "value": "4",
        "desc": "PN9"
      }
    ],
    "min": "0",
    "max": "4",
    "default": "0",
    "value": "0"
  },
  "digital_gain": {
    "type": "int",
    "min": "256",
    "max": "65535",
    "step": "1",
    "default": "256",
    "value": "256"
  }
}

danvandermeir, 2022-02-11:
Thanks, will have to test tomorrow.

The-EG, 2022-02-17:
Hey @danvandermeir , would you be willing to share how you got this camera working with OctoPrint? There's a user on the OctoPrint forums having an issue with an imX519 and I was wondering if it's similar: https://community.octoprint.org/t/request-support-imx519-arcucam-16mpix/42088

danvandermeir, 2022-02-17:
I sent a message over there, I'm stuck at the same spot as everyone else currently. That's also why I haven't updated here.

commit 9cd3811a (2022-02-11): ✨ Add controls for imx559 Arducam AutoFocus HiRes

Fix #69
closed 2022-03-20 by The-EG