The-EG/OctoPrint-CameraSettings #72: [BUG] "White Balance Auto" and "Focus Auto" settings are not applied

Jilocasin, 2022-04-14:
**Observed behavior**
The "White Balance Auto" and "Focus Auto" settings are not stored (or loaded) correctly. The checkbox is always set to disabled when loading from the preset.

The log seems to show some errors here:
```
2022-04-14 00:30:31,563 - octoprint.plugins.camerasettings - INFO - Loading Jilo preset, applying 1 time(s)
2022-04-14 00:30:32,055 - octoprint.plugins.camerasettings - WARNING - Error setting exposure_absolute:
exposure_absolute: Invalid or incomplete multibyte or wide character
VIDIOC_QUERYCAP: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid or incomplete multibyte or wide character

2022-04-14 00:30:32,369 - octoprint.plugins.camerasettings - WARNING - Error setting focus_absolute:
focus_absolute: Invalid or incomplete multibyte or wide character
VIDIOC_QUERYCAP: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid or incomplete multibyte or wide character

2022-04-14 00:30:33,036 - octoprint.plugins.camerasettings - WARNING - Error setting white_balance_temperature:
white_balance_temperature: Invalid or incomplete multibyte or wide character
VIDIOC_QUERYCAP: ok
VIDIOC_S_CTRL: failed: Invalid or incomplete multibyte or wide character
```

Ironically those settings seen in the log seem to be updated correctly. It's just the corresponding "auto" checkboxes that are reset disabled.

**Expected behavior**
The checkboxes should change to the state saved in the preset.

**OctoPrint Version**
1.7.3

**CameraSettings Plugin Version**
0.3.2

**Camera**
Logitech HD Pro C920

**Linux/OctoPi**
0.18.0, running on Raspberry Pi 3 Model B Plus Rev 1.3

**Other Plugins**
Nothing camera related.

The-EG, 2022-04-15:
And these work properly otherwise? Does incrementing the preset load count have any effect?

Jilocasin, 2022-04-15:
I haven't ever played with the exposure settings because the auto exposure seems to work just fine.

When restarting or when applying the preset manually the "Focus auto" checkbox is **reset** . The "Focus absolute" slider is also **reset** to 0. The "White Balance Auto" checkbox is **reset** as well, although **not** the "White Balance Temperature" slider.
I managed to successfully save a value to the color temperature, which is in fact loaded at startup or manual preset load.

Also I noticed that when disabling the "Focus Auto" checkbox, the "Focus absolute" slider is immediately reset to 0 internally when read back by the settings panel.

I tried experimenting with the Preset Apply Count from 1 to 3, but it did not make any difference.
Is there a convenient way to download the saved preset and check the actual stored values?

The-EG, 2022-04-15:
> Is there a convenient way to download the saved preset and check the actual stored values?

The presets are stored in OctoPrint's config.yaml as a plugin setting.

>This makes me think whether it may have something to do with the order of settings that are applied, since they internally depend on each other.

This is likely and exactly why the startup preset apply count exists in the first place : https://github.com/The-EG/OctoPrint-CameraSettings/blob/main/docs/faq.md#not-all-settings-are-applied-on-on-startup

> Some settings can only be set after other settings meet certain conditions. For example, on some cameras exposure time can only be set if exposure auto is off. Unfortunately, the plugin has no way of knowing these dependecies, but it is possible to load a preset multiple times on startup to work around this.

I'm a bit skeptical that you are the first of many users that use this popular camera model to find this issue, but it is possible that no one else has ever touched these settings I guess..

trulow, 2022-05-07:
I just got into using Octoprint and I too am experiencing this issue. Upon restarting my server, I find that I have to continuously go back into the camera settings to "reenable" the auto white balance as it's never activates on it's own even though I saved it to a preset and have "Load Preset on OctoPrint Startup" enabled.

Jilocasin, 2022-05-07:
I saved a preset with auto white balance and focus. The config now includes:
```camerasettings:
load_preset_on_startup: true
multicam_mapping:
- camera: HD Pro Webcam C920
  multicam: null
presets:
- camera: video0
  controls:
	backlight_compensation: '0'
	brightness: '128'
	contrast: '128'
	exposure_absolute: '100'
	exposure_auto: '3'
	exposure_auto_priority: false
	focus_absolute: '0'
	focus_auto: true
	gain: '0'
	pan_absolute: '0'
	power_line_frequency: '1'
	saturation: '128'
	sharpness: '128'
	tilt_absolute: '0'
	white_balance_temperature: '2791'
	white_balance_temperature_auto: true
	zoom_absolute: '338'
  name: Jilo
startup_preset_apply_count: '3'
startup_preset_name: Jilo```

Both settings appear to be saved correctly. So I assume it's something about applying the flags correctly.
I understand how settings may depend on each other when given to v4l2-ctl, but applying the preset multiple times did not change the final settings result.

The-EG, 2022-05-08:
> You can also create a debug entry for this plugin in the OctoPrint logging settings and you'll get lots of debug output, including a line for each setting as it attempts to set it.

@trulow and @Jilocasin please do what I've described above and attach your octoprint.log here.

The-EG, 2022-06-11:
can reopen when more info becomes available.

closed 2022-06-11 by The-EG