The-EG/OctoPrint-CameraSettings #73: Can't add plugin

imawizzard, 2022-04-24:
Hello,

I can't find the plugin in the plugin manager.
When I download the zip file and upload it, it doesn't show up in the plugins.

[image]

The-EG, 2022-04-24:
The plugin manager won't show it if it's not compatible with your setup and even if you manually install it, it still won't work. It requires Python 3 and Linux.

closed 2022-04-24 by The-EG

imawizzard, 2022-04-24:
I used the plugin in the past, why should it not work with my setup? i only
have multicam

The-EG, 2022-04-24:
Well, you've told me absolutely nothing about your setup so I can only guess. If you have filled out the bug report template instead of deleting it or starting with a fresh issue I could have some idea...

so, lets start there:

> **Observed behavior**
> What is happening or not working?
> 
> **Expected behavior**
> What should be happening?
> 
> **OctoPrint Version**
> What version of OctoPrint are you using?
> 
> **CameraSettings Plugin Version**
> What version of this plugin are you using?
> 
> **Camera**
> What camera are you using?
> 
> **Linux/OctoPi**
> What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
> 
> **Other Plugins**
> Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
> 
> 
>

imawizzard, 2022-04-24:
Observed behavior
    Can't install the Camera Settings Plugin

    Expected behavior
   Install the plugin

    OctoPrint Version
    What version of OctoPrint are you using?
1.7.3
    CameraSettings Plugin Version
    What version of this plugin are you using?
Latest (0.3.2)
    Camera
    What camera are you using?
Trust Trino 720P, Endoscope
    Linux/OctoPi
    What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
0.18.0
    Other Plugins
    Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
[image]
[image]

imawizzard, 2022-04-24:
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/8550339/octoprint.log)

The-EG, 2022-04-25:
from your octoprint.log:
> 2022-04-24 19:41:41,660 - octoprint.plugin.core - INFO - 33 plugin(s) registered with the system:
> ...
> | !Camera Settings (0.3.0) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_CameraSettings
> ...
> Prefix legend: ! = disabled, # = blacklisted, * = incompatible

It's installed, you have it disabled.

imawizzard, 2022-04-25:
Didn't know that. I Will check it.

Thank you