The-EG/OctoPrint-CameraSettings #74: [FR] Implement Controls for [imx519 manual focus slide is gone after update]

Lonewolf2nd, 2022-06-11:
**Arducam autofocus 16mp**:

after update, note I don't update everytime, but 3 months ago I installed this plugin and had a nice focus slider, which worked great to focus my camera.
After the update I have many options, but non seem to work or do anything and my camera is now out of focus.
Downgrading didn't help either, only getting a notification that my camera isn't supported.

{
  "camera_orientation": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Front"
      },
      {
        "value": "1",
        "desc": "Back"
      },
      {
        "value": "2",
        "desc": "External"
      }
    ],
    "min": "0",
    "max": "2",
    "default": "2",
    "value": "2",
    "flags": "read"
  },
  "camera_sensor_rotation": {
    "type": "int",
    "min": "0",
    "max": "0",
    "step": "1",
    "default": "0",
    "value": "0",
    "flags": "read"
  },
  "vertical_blanking": {
    "type": "int",
    "min": "100",
    "max": "8383280",
    "step": "1",
    "default": "100",
    "value": "2560"
  },
  "horizontal_blanking": {
    "type": "int",
    "min": "5691",
    "max": "5691",
    "step": "1",
    "default": "5691",
    "value": "5691",
    "flags": "read"
  },
  "analogue_gain": {
    "type": "int",
    "min": "0",
    "max": "960",
    "step": "1",
    "default": "0",
    "value": "771"
  },
  "red_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "green_red_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "blue_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "green_blue_pixel_value": {
    "type": "int",
    "min": "0",
    "max": "4095",
    "step": "1",
    "default": "4095",
    "value": "4095"
  },
  "pixel_rate": {
    "type": "int64",
    "min": "686000000",
    "max": "686000000",
    "step": "1",
    "default": "686000000",
    "value": "686000000",
    "flags": "read"
  },
  "test_pattern": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Disabled"
      },
      {
        "value": "1",
        "desc": "Color Bars"
      },
      {
        "value": "2",
        "desc": "Solid Color"
      },
      {
        "value": "3",
        "desc": "Grey Color Bars"
      },
      {
        "value": "4",
        "desc": "PN9"
      }
    ],
    "min": "0",
    "max": "4",
    "default": "0",
    "value": "0"
  },
  "digital_gain": {
    "type": "int",
    "min": "256",
    "max": "65535",
    "step": "1",
    "default": "256",
    "value": "256"
  }
}

hopefully you can help.

The-EG, 2022-06-11:
Hi there, recent updates to this plugin shouldn't have changed anything for
that camera. The only thing that comes to mind is that relative controls
are now shown as buttons instead of siders.

Can you take a screenshot of what controls you currently get?

Also, please add a debug entry for this plugin in the OctoPrint logging
settings, close the settings window and then go back to the camera settings
page and then attach your OctoPrint.log here. That file can be downloaded
from the logging settings page.

The-EG, 2022-06-11:
Also, if you wanted to try to downgrade to the last release, it should have support for the imx519, but not the new relative control changes.

Just install with this URL: https://github.com/The-EG/OctoPrint-CameraSettings/archive/refs/tags/0.3.2.zip

Lonewolf2nd, 2022-06-11:
Hi,

Thanks for the reply. I've tried version 0.3.0 and 0.3.1, but going to try
tomorrow after my print also 0.3.2 as you suggested. in your other reply.
attached some screenshots and a picture from a website how it used to look
like, but it shows a different camera.
also the website link where I got the picture from, as that was part of the
setup I used a few months back.
https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide/autofocus-camera-solution-using-on-octopi/

Also attached how it looked like when it was in focus. this is a reddit
link of my first octolase with it. note: this resolution is higher, due It
are pictures and not a video stream.
https://www.reddit.com/r/octoprint/comments/t62w75/my_first_octolapse_with_a_arducam_16mp_autofocus/

I've added the camera settings plugin to the log and tried some buttons and
sliders, which won't change anything in the preview and also nothing in my
control view windows.
Hopefully it helps.

Many thanks already

Pim

Lonewolf2nd, 2022-06-12:
Hi,

I've now also reinstall the kernel drivers again.

wget -O install_pivariety_pkgs.sh
https://github.com/ArduCAM/Arducam-Pivariety-V4L2-Driver/releases/download/install_script/install_pivariety_pkgs.sh

chmod +x install_pivariety_pkgs.sh

./install_pivariety_pkgs.sh -p imx519_kernel_driver

Now there appears an 'ak7375 10-000c' camera option and this gives me the
focus slider.
This one made my camera in focus again.

I guess it isn't how it should work, but it works :).

The-EG, 2022-06-13:
Ok, so is it working with 0.4.0 now or just 0.3.2?

The-EG, 2022-07-18:
Closing due to inactivity.

closed 2022-07-18 by The-EG

Lonewolf2nd, 2022-10-11:
Got it working under 0.3.2 and than updated in octoprint to.0.4.0 and it
still worked.