The-EG/OctoPrint-CameraSettings #75: Autofocus and focus absolute is not saved[BUG] New Issue

wishmaster86, 2022-06-13:
**Observed behavior**
I have disabled autofocus and configured Focus absolute (81) then saved these settings. 2 times because autofocus needs a save first and then focus absolute.

After a reboot or shutdown of the Pi and the Pi starts again the autofocus and absolute settings are not restored.

**Expected behavior**
Autofocus to be off and Focus absolute to be configured as 81.

**OctoPrint Version**
What version of OctoPrint are you using?
0.18.0

**CameraSettings Plugin Version**
What version of this plugin are you using?
0.3.2

**Camera**
What camera are you using?
HP Webcam HD 3100 720p

**Linux/OctoPi**
What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
Octopi 1.8.1

**Other Plugins**
Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
No. Vanilla install.

The-EG, 2022-06-13:
Hi there, have you increased the `Startup Preset Apply Count` to 2 or 3 and selected the appropriate preset under 'Startup Preset'?

mthayne, 2022-06-23:
I have the same problem. I've tried raising the apply count to as high as 4. And I, as far as I can see, are using the appropriate startup preset. It seems to be forgetting both the auto focus setting and the manual focus value.

The-EG, 2022-07-18:
Either of you: please enable debugging logging for this plugin (OctoPrint settings, Logging), reproduce the behavior (ie. restart) and then upload your octoprint.log here.

The-EG, 2022-08-10:
Closing due to inactivity.

closed 2022-08-10 by The-EG

mthayne, 2022-08-12:
Hi,
I missed your earlier emails. I've got a log if it would still help.
Thanks.