The-EG/OctoPrint-CameraSettings #76: [FR] Implement Controls for [Camera Name]

doctorevil30564, 2022-06-16:
**Camera manufacturer and model Logitech C920
#Note this is running on a Orange Pi 3 LTS, same camera has been working perfectly on a raspberry pi 3B+ that has started having constant issues with dropping wifi connections. I followed the manual install process for a Raspberry Pi to install Octoprint on the Orange Pi 3 LTS

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "hevc_slice_parameters": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  },
  "hevc_decode_mode": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Slice-Based"
      }
    ],
    "min": "0",
    "max": "0",
    "default": "0",
    "value": "0"
  },
  "hevc_start_code": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "No Start Code"
      }
    ],
    "min": "0",
    "max": "0",
    "default": "0",
    "value": "0"
  },
  "h264_decode_mode": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "Slice-Based"
      }
    ],
    "min": "0",
    "max": "0",
    "default": "0",
    "value": "0"
  },
  "h264_start_code": {
    "type": "menu",
    "values": [
      {
        "value": "0",
        "desc": "No Start Code"
      }
    ],
    "min": "0",
    "max": "0",
    "default": "0",
    "value": "0"
  },
  "h264_sequence_parameter_set": {
    "type": "unknown"
  },
  "h264_picture_parameter_set": {
    "type": "unknown"
  },
  "h264_scaling_matrix": {
    "type": "unknown"
  },
  "h264_prediction_weight_table": {
    "type": "unknown"
  },
  "h264_slice_parameters": {
    "type": "unknown"
  },
  "h264_decode_parameters": {
    "type": "unknown"
  },
  "vp8_frame_parameters": {
    "type": "unknown"
  },
  "mpeg_2_sequence_header": {
    "type": "unknown"
  },
  "mpeg_2_picture_header": {
    "type": "unknown"
  },
  "mpeg_2_quantisation_matrices": {
    "type": "unknown"
  }
}

doctorevil30564, 2022-06-16:
getting this info for the camera model

pi@orangepi3-lts:~$ v4l2-ctl -D
Driver Info:
        Driver name      : uvcvideo
        Card type        : HD Pro Webcam C920
        Bus info         : usb-5311000.usb-1
        Driver version   : 5.16.17
        Capabilities     : 0x84a00001
                Video Capture
                Metadata Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps      : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Media Driver Info:
        Driver name      : uvcvideo
        Model            : HD Pro Webcam C920
        Serial           : C2EC3A6F
        Bus info         : usb-5311000.usb-1
        Media version    : 5.16.17
        Hardware revision: 0x00000011 (17)
        Driver version   : 5.16.17
Interface Info:
        ID               : 0x03000002
        Type             : V4L Video
Entity Info:
        ID               : 0x00000001 (1)
        Name             : HD Pro Webcam C920
        Function         : V4L2 I/O
        Flags         : default
        Pad 0x01000007   : 0: Sink
          Link 0x0200001f: from remote pad 0x100000a of entity 'Processing 3': Data, Enabled, Immutable

getting this info for the available controls

pi@orangepi3-lts:~$ v4l2-ctl -L

User Controls

                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
        white_balance_automatic 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=44
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
                                0: Disabled
                                1: 50 Hz
                                2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=6500 step=1 default=4000 value=6500 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0

Camera Controls

                  auto_exposure 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
                                1: Manual Mode
                                3: Aperture Priority Mode
         exposure_time_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=333 flags=inactive
     exposure_dynamic_framerate 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=0
                 focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=0 flags=inactive
     focus_automatic_continuous 0x009a090c (bool)   : default=1 value=1
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=100 value=100

The-EG, 2022-06-16:
That's interesting that it's coming back with more controls. 
The good thing is that none of those are really that useful, so you can continue to use the plugin. I'll add those so that the warning goes away, though.
Thanks!

The-EG, 2022-08-10:
Sorry for the delay, this is now available in 0.4.1rc1. You can try this version out by switching to the 'Release Candidate' channel for this plugin in the Software Updates section of OctoPrint's settings.

doctorevil30564, 2022-08-10:
Awesome, thank you.

commit f8716921 (2022-08-10): ✨ Add camera controls

Fixes #79 (frame_rate)
Fixes #78 (min_number_of_capture_buffers)
Fixes #76 & Fixes #70 (h264 decode mode/start and hevc decode/start)
closed 2022-11-20 by The-EG