The-EG/OctoPrint-CameraSettings #77: Camera settings lost any time the screen reloads

timropp, 2022-06-19:
Octoprint 1.8.1 running on RPi4. Camera is Logitech C920. 

Camera settings works great to set and save a preset. However, whenever the screen reloads (either reload the webpage, or use an Octoprint app on my phone), it does not load the preset that it's supposed to. Instead, it load some bizarre settings that are NOT default to the camera or any preset, but instead are almost completely dark. I have to go into the plugin settings and reload my preset, and then it works great until something causes the page to reload again.

The-EG, 2022-07-18:
Please enable debugging logging for this plugin (OctoPrint Settings, logging), reproduce that behavior and then upload your octoprint.log here (also available on the logging settings page).

timropp, 2022-07-18:
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/9136149/octoprint.log)

I booted up octoprint, and the camera was nearly black. I have the preset set to load on startup but it doesn't. I went into the camera settings plugin and manually loaded the preset. Looks great. Closed out of the settings and reloaded my Chrome page. The camera comes up as black again, so I manually reloaded the preset again.

The-EG, 2022-07-24:
The plugin only reports changing the camera settings twice:
```log
2022-07-18 18:49:24,752 - octoprint.plugins.camerasettings - DEBUG - Controls: ['backlight_compensation=0', 'brightness=140', 'contrast=160', 'exposure_absolute=312', 'exposure_auto=1', 'exposure_auto_priority=False', 'focus_absolute=100', 'focus_auto=False', 'gain=156', 'led1_frequency=0', 'led1_mode=0', 'pan_absolute=0', 'power_line_frequency=2', 'saturation=128', 'sharpness=128', 'tilt_absolute=0', 'white_balance_temperature=5000', 'white_balance_temperature_auto=False', 'zoom_absolute=100']
2022-07-18 18:49:43,722 - octoprint.plugins.camerasettings - DEBUG - Controls: ['backlight_compensation=0', 'brightness=140', 'contrast=160', 'exposure_absolute=312', 'exposure_auto=1', 'exposure_auto_priority=False', 'focus_absolute=100', 'focus_auto=False', 'gain=156', 'led1_frequency=0', 'led1_mode=0', 'pan_absolute=0', 'power_line_frequency=2', 'saturation=128', 'sharpness=128', 'tilt_absolute=0', 'white_balance_temperature=5000', 'white_balance_temperature_auto=False', 'zoom_absolute=100']
```

Both times to the exact same settings when you load a preset. So, it's not the plugin that is changing the camera. I see you have OctoLapse installed but disabled. Did you ever have Custom Image Preferences enabled? Even with the plugin itself disabled, that could be interfering and setting it's own settings.

Another way to check would be to change your `camera_http_options` in octopi.txt back to `-n`:
```ini
camera_http_options="-n"
```

That would keep anything that's using mjpg-streamers web interface from changing the camera.

timropp, 2022-07-24:
Thanks for the suggestions. Sadly, no luck. I re-enabled octolapse and made sure that custom image preferences was not on - it's not. Then I disabled octolapse and even uninstalled it.
I made the settings change in octopi.txt. 
Reboot, and still the same thing. 
Then I uninstalled the camera settings plugin and rebooted the pi. Image came up bright. I reinstalled camera settings, recreated the preset, and it looked good. But then I reloaded the page and bam, black again. So I have no idea at this point why, but it's something between the camera/pi/possibly something else installed/this plugin.

The-EG, 2023-07-02:
Can't reproduce this or do anything useful with it. Closing.

closed 2023-07-02 by The-EG