The-EG/OctoPrint-CameraSettings #80: [FR] REST Api interface

nfedyashev, 2022-08-11:
**Description of Desired Feature**

CameraSettings works great using web interface but sometimes I miss REST HTTP interface that can be used for controlling it programmatically. Basically, this is for snapshots automation.

So far I'm only using focus/pan/zoom features.

> Does another plugin already provide this feature?  
I don't think so, at least I wasn't able to find one.

From what I found, some plugins seem to have API REST interface so it's doable. I'm just not sure how difficult this might be because I'm not a Python developer.

The-EG, 2022-08-12:
Hi there, there are a few simple api methods that the plugin uses internally already, they just aren't documented. Based on what you've described, what you need is already available.

You can set camera controls via the plugins simple api already via the `set_camera_controls` command at `/api/plugin/camerasettings`. For example:
```terminal
$ curl -H "X-Api-Key: (APIKEY)" -H "Content-Type: application/json" -d '{"command":"set_camera_controls", "camera":"video0", "controls":{"brightness":50}}' http://octopi.local/api/plugin/camerasettings
```

Note it is using the 'internal' names for everything, ie you need to know the video device (`video0` in this case), the actual control name (as shown by `v4l2-ctl --list-ctrls-menus`) and what values to set. If it's a menu control, you need the value, not the description (`0` instead of `disabled`). 

Also, all this plugin is doing is running `v4l2-ctl` to make the changes. If you are trying to script something on the same device that the camera is running on, you could also do the same just as easily.

closed 2022-08-12 by The-EG

nfedyashev, 2022-08-12:
that is awesome! Thank you so much.

In case anyone has a similar task and found this issue using Github search or Google, below is what I used for this task:

```

#+BEGIN_SRC http
POST http://octoprint.local/api/plugin/camerasettings
Content-Type: application/json
Authorization: Bearer ${access-token}

{
  "camera": "video0",
  "command": "set_camera_controls",
  "controls": {
    "focus_absolute": "9",
    "zoom_absolute": "90",
    "pan_absolute": "-21600"
  }
}
#+END_SRC
```

control names can be seen in Chrome web inspector each time you update something, or like Taylor explained above using v4l2-ctl command

For Logitech Webcam HD Pro C920 it gave me the following ones:
```
v4l2-ctl --list-ctrls-menus
                     brightness 0x00980900 (int)    : min=0 max=255 step=1 default=128 value=128
                       contrast 0x00980901 (int)    : min=0 max=255 step=1 default=128 value=128
                     saturation 0x00980902 (int)    : min=0 max=255 step=1 default=128 value=128
 white_balance_temperature_auto 0x0098090c (bool)   : default=1 value=1
                           gain 0x00980913 (int)    : min=0 max=255 step=1 default=0 value=50
           power_line_frequency 0x00980918 (menu)   : min=0 max=2 default=2 value=2
				0: Disabled
				1: 50 Hz
				2: 60 Hz
      white_balance_temperature 0x0098091a (int)    : min=2000 max=6500 step=1 default=4000 value=5838 flags=inactive
                      sharpness 0x0098091b (int)    : min=0 max=255 step=1 default=128 value=128
         backlight_compensation 0x0098091c (int)    : min=0 max=1 step=1 default=0 value=0
                  exposure_auto 0x009a0901 (menu)   : min=0 max=3 default=3 value=3
				1: Manual Mode
				3: Aperture Priority Mode
              exposure_absolute 0x009a0902 (int)    : min=3 max=2047 step=1 default=250 value=333 flags=inactive
         exposure_auto_priority 0x009a0903 (bool)   : default=0 value=1
                   pan_absolute 0x009a0908 (int)    : min=-36000 max=36000 step=3600 default=0 value=-21600
                  tilt_absolute 0x009a0909 (int)    : min=-36000 max=36000 step=3600 default=0 value=32400
                 focus_absolute 0x009a090a (int)    : min=0 max=250 step=5 default=0 value=10
                     focus_auto 0x009a090c (bool)   : default=1 value=0
                  zoom_absolute 0x009a090d (int)    : min=100 max=500 step=1 default=100 value=100
                      led1_mode 0x0a046d05 (menu)   : min=0 max=3 default=3 value=3
				0: Off
				1: On
				2: Blink
				3: Auto
                 led1_frequency 0x0a046d06 (int)    : min=0 max=255 step=1 default=0 value=0

```