The-EG/OctoPrint-CameraSettings #83: [BUG] manual focus does not load properly

th3d00d, 2022-09-07:
I've saved numerous profiles with a set fixed focus of "300". I also set one of those profiles to load on startup. Whenever I turn off the Octopi and turn it back on, the focus is set to "180". I've tried deleting all profiles and creating new ones, I also tried loading the profile manually. the focus does not seem to get saved/loaded.

I am using an USB Webcam with autofocus, but I disabled autofocus in the plugin. if I manually set the focus to "300" after startup, it stays that way until the octopi is shut down.

I am running OctoPrint 1.8.2 with the CameraSettings Plugin version 0.4.0. My OctoPi version is 0.18.0.

the only other Camera Plugin I am running is Octolapse, but I don't think this will change the focus in any way, as it does not change once I set it manually.

The-EG, 2022-09-08:
Hi there,
Have you increased the `Startup Preset Apply Count` to 2 or 3 as described in the [FAQ](https://github.com/The-EG/OctoPrint-CameraSettings/blob/main/docs/faq.md#not-all-settings-are-applied-on-on-startup)?

If so and it's still happening, please create a debug entry for this plugin:

- Open OctoPrint Settings
- Goto OctoPrint->Logging
- Under Logging Levels and Name, Change `Select a logger` to octoprint.plugins.camerasettings
- Click save

After that restart OctoPrint or the entire RPi. Then go back to the same page in OctoPrint settings, click the download icon for `octoprint.log` under Action and attach that file here. After that you can delete the debug logging line for this plugin below.

th3d00d, 2022-09-14:
Hello, I've not been able to try anything new so far, but I tried increasing the startup Preset Apply Count up to 10, which did not make any difference. I will try again in a few days and report back.

th3d00d, 2023-01-05:
Very sorry about the delay, but when finally starting up my octoprint with the autofocus camera again, the issue remained - I attached my log to this post for you:
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/10353260/octoprint.log)

The-EG, 2023-03-17:
@th3d00d Creating duplicate issues on purpose is annoying and a great way to motivate developers to *NOT* help you. You could have simply commented here instead.

Your log would indicate that your preset only contained 3 values:
```
2023-01-05 16:10:06,552 - octoprint.plugins.camerasettings - DEBUG - Setting controls on video0
2023-01-05 16:10:06,574 - octoprint.plugins.camerasettings - DEBUG - Controls: ['brightness=114', 'contrast=106', 'saturation=90']
...
023-01-05 16:10:06,601 - octoprint.plugins.camerasettings - DEBUG - Setting brightness = 114 on video0
...
2023-01-05 16:10:06,753 - octoprint.plugins.camerasettings - DEBUG - v4l2-ctl ran successfully:
...
2023-01-05 16:10:06,753 - octoprint.plugins.camerasettings - DEBUG - Setting contrast = 106 on video0
...
2023-01-05 16:10:06,867 - octoprint.plugins.camerasettings - DEBUG - v4l2-ctl ran successfully:
...
2023-01-05 16:10:06,908 - octoprint.plugins.camerasettings - DEBUG - Setting saturation = 90 on video0
...
2023-01-05 16:10:07,007 - octoprint.plugins.camerasettings - DEBUG - v4l2-ctl ran successfully:
```

Do you see any errors in your browser console (ctrl-shift-I) when saving the preset?

As is, I can't reproduce this and there are no actual errors.

th3d00d, 2023-03-17:
I am very sorry, my comment on this thread seemed to go unnoticed, I assumed the notifications for this issue must have been turned off somehow.

when checking the console and saving a new preset, there is no message showing up. when clicking on "save" on the plugin window itself (not the preset), it shows "ConfiguredLoggers has not changed. Not saving."

The-EG, 2023-03-17:
Ok, I'll have to add some additional debugging to hopefully see why you aren't saving the preset properly. In your case it would appear it's never being saved, so it's never being loaded.

th3d00d, 2023-03-17:
That would make sense, Thank you for the fast response :)

th3d00d, 2023-05-19:
Did you add the debugging to catch the bug so far?
I noticed the profiles are being saved in the YAML file, but they just don't seem to load properly

The-EG, 2023-05-20:
If it's saving it in the YAML then there's not much else to verify other than what the debugging output already has. It's really pretty simple, it loads the settings as a dictionary, applies them one by one (possibly in a loop if multiple startup applies are selected). If you see it in YAML but not the debug output then something impossible is happening. That would indicate that either it's not in the YAML after all or it is actually in the DEBUG output somewhere. This plugin doesn't handle loading the configuration (YAML), so if there were some problem there, all plugins would be affected.

th3d00d, 2023-05-20:
I noticed, if I put the commands into a startup script they seem to get overwritten by something. If I use a SSH Terminal to send the command after startup (like, half a minute at most) - it works just fine, just as if I would select the settings again in your plugin.
Could you possibly add an option to delay the loading of the profiles by a set amount of time? I think that could fix the issue, without having to go through every single possible other cause

The-EG, 2023-05-20:
Not really, I have no way of knowing when OctoPrint is being loaded (and hence the plugin) and I can't hold up loading while the plugin is being loaded because that would in turn hold up OctoPrint from loading. I could offload all that to a thread, make a timer, etc. but that's A LOT of complexity to add just to solve something that isn't even related to this plugin or even OctoPrint if that is the case.

The-EG, 2023-05-20:
If that is the case it'd be easier to just cause OctoPrint to load later

th3d00d, 2023-05-20:
I meant, just like having the plugin reload the set profile after a delay instead (or in addition to) directly at startup

The-EG, 2023-05-20:
There is no 'after a delay' unless I implement all that myself. When OctoPrint is loading plugins its doing so in a single thread, so I can't just insert a delay there.

th3d00d, 2023-05-20:
Yeah, this is what I meant, as a feature request that could avoid potential issues, just a timer to load the profile again after, lets say, 30 seconds

I always shut the Raspberry Pi down completely and disconnect the power when not in use, so the issue happens for me when I am starting the device from a total power off state

The-EG, 2023-07-06:
Closing as this isn't an actual issue with the plugin but something else at system startup.

closed 2023-07-06 by The-EG