The-EG/OctoPrint-CameraSettings #85: [BUG] White Balance Auto unchecked every time I restart octoprint

TarcisioMenezes, 2022-09-21:
**Observed behavior**
What is happening or not working?
White Balance Auto unchecked every time I restart octoprint

[image]

**Expected behavior**
What should be happening?
It should be on since that was the way I saved the settings. 
**OctoPrint Version**
What version of OctoPrint are you using?
Version 1.8.3
**CameraSettings Plugin Version**
What version of this plugin are you using?
0.4.0
**Camera**
What camera are you using?
Logitech C920x
**Linux/OctoPi**
What Linux distribution/image is this running on? If you are unsure, it is probably OctoPi. If OctoPi, what version?
Octopi Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.4
**Other Plugins**
Do you have any other plugins that may be modifying the camera stream (ie, The Spaghetti Detective)?
No.

The-EG, 2022-09-22:
Hi there, I also saw and responded to your comments in Discord. Feel free to contact me there too if a quick back/forth is easier.

Do you have OctoLapse installed with Custom Image Preferences enabled? That can interfere with this plugin.

Also, can you please upload a debug log:
 1. Goto 'Logging' in OctoPrint settings.
 2. Create a DEBUG entry for this plugin (octoprint.plugins.camerasettings) in the lower part of the window.
 3. Click save
 4. Reproduce your issue (reboot/restart)
 5. Go back to the Logging settings, download `octoprint.log` and post it here
 6. Delete the DEBUG entry you created above

The-EG, 2022-09-22:
Just a note from our discussion on Discord:
 user is using HLS instead of MJPG, via ffmpeg. Probably not related, but I'm going to do some testing to rule it out. I'm curious if this is related to a few other similar issues I've seen recently.

TarcisioMenezes, 2022-09-22:
here we go
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/9627893/octoprint.log)

TarcisioMenezes, 2022-09-22:
Here we go
[octoprint.log](https://github.com/The-EG/OctoPrint-CameraSettings/files/9628027/octoprint.log)

acevalsl, 2022-10-13:
Hello. Same problem, the automatic white balance is disconnected every time I turn on. Latest version of octoprint and latest version of pi

The-EG, 2022-10-14:
@acevalsl Are you also using HLS?

acevalsl, 2022-10-14:
What is HLS? I'm not using any camera plugins but I'm not sure what you're asking me

jimmiedave, 2023-01-09:
Same issue here, not using HLS (at least haven't set it up - it's a plain vanilla logitech usb cam and a plain vanilla setup (when this started - have started using multicam since, but both before and after, camera settings wouldn't persist the "auto white balance" checkbox).

The-EG, 2023-01-09:
> Questioning whether this is something messing with the camera settings: wouldn't the AWB setting/checkmark be persisted in the Camera Settings file regardless of what other software was doing to the Logi camera?

Yes, and you can check this yourself. There is no 'camera settings file,' though, everything is stored in the OctoPrint settings, config.yaml.

But I have verified this by the log provided above. The setting is there and it is clearly being set on server startup:
```
2022-09-22 11:51:26,737 - octoprint.plugins.camerasettings - DEBUG - Setting white_balance_temperature_auto = True on video0
2022-09-22 11:51:26,795 - octoprint.plugins.camerasettings - DEBUG - v4l2-ctl ran successfully:
VIDIOC_QUERYCAP: ok
VIDIOC_S_CTRL: ok
```

So, from the perspective of this plugin, everything should be working properly. I can't reproduce this, nor get any information in the log saying it's even happening. The only logical explanation left is that either it's not happening (not likely with multiple users) or you all have a similar setup with something tweaking settings in the background.

bakerlogan, 2023-02-13:
Hi, having a similar issue where settings aren't loading, auto white balance and absolute focus are the two I care about. 

I'm using a NexiGo N930E FHD webcam. 

Below are the camera setting related messages I see in the Octoprint log. I just turned on debug logging so I'll share more if I see anything relevant, but apparently there are some invalid arguments when it tries to set a few of the settings. 

```
2023-02-12 01:42:41,294 - octoprint.plugins.camerasettings - INFO - Loading Focus-185, Auto balance preset, applying 1 time(s)
2023-02-12 01:42:41,335 - octoprint.plugins.gpiocontrol - INFO - Configured GPIO26: active_high,default_off (Light)
2023-02-12 01:42:41,336 - octoprint.plugins.uicustomizer - INFO - UI Customizer is initialized.
2023-02-12 01:42:41,336 - octoprint.plugins.wemoswitch - INFO - WemoSwitch loaded!
2023-02-12 01:42:41,413 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2023-02-12 01:42:41,491 - octoprint.plugins.camerasettings - WARNING - Error setting exposure_absolute:
exposure_absolute: Invalid argument
VIDIOC_QUERYCAP: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid argument

2023-02-12 01:42:41,662 - octoprint.plugins.camerasettings - WARNING - Error setting focus_absolute:
focus_absolute: Invalid argument
VIDIOC_QUERYCAP: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid argument

2023-02-12 01:42:41,862 - octoprint.plugins.camerasettings - WARNING - Error setting white_balance_temperature:
white_balance_temperature: Invalid argument
VIDIOC_QUERYCAP: ok
VIDIOC_S_CTRL: failed: Invalid argument

2023-02-12 05:35:42,102 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture0 based on ^bcm2835-isp-\w+$
2023-02-12 05:35:42,105 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-isp based on ^bcm2835-codec-\w+$
2023-02-12 05:35:42,132 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-decode based on ^bcm2835-codec-\w+$
2023-02-12 05:35:42,135 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-capture1 based on ^bcm2835-isp-\w+$
2023-02-12 05:35:42,137 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-output0 based on ^bcm2835-isp-\w+$
2023-02-12 05:35:42,139 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-codec-encode based on ^bcm2835-codec-\w+$
2023-02-12 05:35:42,173 - octoprint.plugins.camerasettings - INFO - Excluding camera bcm2835-isp-stats2 based on ^bcm2835-isp-\w+$
2023-02-12 05:35:50,539 - octoprint.plugins.camerasettings - WARNING - Error setting exposure_absolute:
exposure_absolute: Invalid argument
VIDIOC_QUERYCAP: ok
VIDIOC_S_EXT_CTRLS: failed: Invalid argument
```

The-EG, 2023-07-02:
I noticed earlier that `True` was being passed instead of `1`:
> 2022-09-22 11:51:26,737 - octoprint.plugins.camerasettings - DEBUG - Setting white_balance_temperature_auto = True on video0
> 2022-09-22 11:51:26,795 - octoprint.plugins.camerasettings - DEBUG - v4l2-ctl ran successfully:
> VIDIOC_QUERYCAP: ok
> VIDIOC_S_CTRL: ok

As you can see, v4l2-ctl doesn't complain about this, but it also doesn't properly set the value. I've fixed this in the latest release, 0.4.3. This should resolve this type of issue, give it a try.

bakerlogan, 2023-07-04:
It no longer seems to be an issue, thanks!

closed 2023-07-05 by The-EG