The-EG/OctoPrint-CameraSettings #87: [FR] Implement Controls for [Camera Name]

alexian79, 2022-10-19:
Hi this is jetson nano camera imx219, Thank you.

**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "group_hold": {
    "type": "bool",
    "default": "0",
    "value": "0",
    "flags": "execute"
  },
  "sensor_mode": {
    "type": "int64",
    "min": "0",
    "max": "6",
    "step": "1",
    "default": "0",
    "value": "0",
    "flags": "slider"
  },
  "frame_rate": {
    "type": "int64",
    "min": "2000000",
    "max": "21000000",
    "step": "1",
    "default": "21000000",
    "value": "21000000",
    "flags": "slider"
  }
}

The-EG, 2022-11-11:
Hi there, sorry for the delay. I plan to add these and then publish a new release this weekend!

alexian79, 2022-11-11:
Thank you, not pushing here, just a humble request :)

commit f6aaa40d (2022-11-20): ✨ Add controls forJetson Nano imx219 (fix #87)
closed 2022-11-20 by The-EG