The-EG/OctoPrint-CameraSettings #106: [FR] Implement Controls for [Pi Camera Module 3]

andrewlockey, 2023-01-26:
**Camera manufacturer and model (ie. NexiGO N660)**: 
Pi Camera Module 3, 12 Megapixel Sony IMX708 image sensor, powered autofocus, improved low-light sensitivity and HDR 

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
}

The-EG, 2023-01-26:
Hmm...These unknown controls should already be hidden. And I've definitely included the new PiCam3 in the latest RC.

What version of this plugin are you using?

achampion, 2023-02-06:
Same for me: Camera Settings (0.4.1)

The-EG, 2023-02-26:
> However, with both 0.4.1 and 0.4.2RC1 nothing is sticking, e.g. clicking off Vertical Flip (which defaulted to on), it just immediately jumps back on.

Can you please see if there is any errors in the browser console (press ctrl-shift-I) and also provide a debug log by adding a debug entry for this plugin in the 'Logging' section of OctoPrint settings, reproducing this behavior, and then uploading your octoprint.log here (also available on that logging settings page).

achampion, 2023-03-12:
Yes, I am getting console areas for any interaction with the camera settings.

I get these 2 errors for any interaction with the camera settings, e.g. attempt to flip horizontal:
[image]
[image]

I also upgraded to 1.9.0RC2 and still no difference.

MarvinLind, 2023-03-19:
This was resolved for me by 0.4.2RC1. I did however not have all of those "Unknown Controls" only "hevc_scaling_matrix" (I think).

The-EG, 2023-03-19:
Thank you, finally someone confirmed! 😄