The-EG/OctoPrint-CameraSettings #115: [FR] Implement Controls for [Camera Name]

Elangwondo, 2023-03-01:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:

Elangwondo, 2023-03-01:
> **Camera manufacturer and model (ie. NexiGO N660)**:
> 
> **Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:

{
  "hevc_sequence_parameter_set": {
    "type": "unknown"
  },
  "hevc_picture_parameter_set": {
    "type": "unknown"
  },
  "slice_param_array": {
    "type": "unknown"
  },
  "hevc_scaling_matrix": {
    "type": "unknown"
  },
  "hevc_decode_parameters": {
    "type": "unknown"
  }
}

The-EG, 2023-03-03:
duplicate of #106

closed 2023-03-03 by The-EG