The-EG/OctoPrint-CameraSettings #128: [FR] Implement Controls for [Camera Name]

danieldhastings, 2023-03-15:
**Camera manufacturer and model (ie. NexiGO N660)**:

**Unknown controls information (click 'Copy to Clipboard' on the warning and paste here)**:
{
  "wide_dynamic_range": {
    "type": "bool",
    "default": "0",
    "value": "0",
    "flags": "grabbed"
  }
}

The-EG, 2023-03-15:
duplicate of #100, this is already available in the latest release candidate.

As far as focus goes, that's only going to be controllable if this v3 module has manual focus. I've heard reports that the controls are split across multiple devices as you've seen. If focus is showing up as a control but isn't having any effect, create a new issue with your octoprint.log attached after first turning on debug logging for this plugin.

closed 2023-03-15 by The-EG

danieldhastings, 2023-03-15:
Updating to the latest release candidate seems to have fixed this, even the autofocus.  The picture is clear and bright and there are no error messages. Thanks for the quick response. Also, thanks for the excellent plugin.